ucb-bar/chiseltest #368: chiseltest 0.5 breaks 0.3 code

schoeberl, 2021-08-14:
ChiselTest 0.5 uses ScalaTest 3.2 where testing classes have been moved. E.g. ```FlatSpec``` is now in ```org.scalatest.flatspec.AnyFlatSpec```. ChiselTest 0.3 uses ScalaTest 3.0 where this class is in ```org.scalatest.FlatSpec```.

If this will be part of the Chisel 3.5 release, this is a *very* bumpy road for ChiselTest users as all their code will break (just for my Chisel book testers these are quite many).

I would like to argue that we really should avoid API changes without a deprecation phase. I guess some version of ScalaTest 3.1.x does allow both versions and depreciate the 3.0 version.

The proposal for this issue is to use a ScalaTest version 3.1.x that deprecated ```FlaSpec``` for the next Chisel release. If this is not possible we need a heavy warning on the release notes what the upgrade to Chisel 3.5 involves. I hope we can avoid that.

closed 2021-08-16 by ekiwi

mwachs5, 2022-01-03:
Curious does bumping to 3.4.4 get you the deprecation phase that you need? My impression is that we would do all necessary deprecations on the stable release branch and allow breaking changes on the major release branch.

schoeberl, 2022-01-03:
I'll give it a try. What is the correct chiseltest version for Chisel 3.4.4

schoeberl, 2022-01-03:
No, it does not.

ekiwi, 2022-01-03:
I think I found the culprit @schoeberl : While we did downgrade `scalatest` for `chiseltest` (thanks to you), the same did not happen for the old `iotesters` which now depends on a `3.2` version of `scalatest`: https://mvnrepository.com/artifact/edu.berkeley.cs/chisel-iotesters_2.13/2.5.0-RC2

Seems like this is the reason why you get the newer version pulled in.

schoeberl, 2022-01-03:
OK, great that you found it!

ekiwi, 2022-01-03:
If you can make a PR to downgrade the `scalatest` dependency to the same one we have for `chiseltest`, I can approve it in no time.

commit 52103ff1 (2021-08-16): Use ScalaTest 3.1 for derpication warnings (#372)

* Use ScalaTest 3.1 for derpication warnings

* add test for deprecated warning

* ScalaTest 3.1.4, fixes #368