ucb-bar/chiseltest #395: Feature request: let users customize which Chisel stages/phases to run when building a test

debs-sifive, 2021-09-09:
It would be nice if [`test()`](https://github.com/ucb-bar/chisel-testers2/blob/ed8cc043a1e45f01a05fe89541e1f68e9bf3c59f/src/main/scala/chiseltest/ChiselScalatestTester.scala#L121) allowed users to specify which Chisel stages/phases they want to run during compilation.

This would allow users to shape their unit test flows to match their actual compilation flows as closely as possible.

It would also be nice if we could insert custom stages/phases in this manner.

ekiwi, 2021-09-09:
Could you provide an example of a custom stage that we could use to mockup a use case scenario?

There are a couple of things I am trying to understand:
- Do stages/phases compose, i.e., will users be able to provide more than one stage or phase?
- Are you expecting to pass command line flags to your stage/phase?
- How do custom stages/phases interact with the various backends supported by chiseltest (Verilator, Treadle, Formal, etc.)?

ekiwi, 2021-09-09:
One potential solution may be to allow users to run their stage and then just provide the resulting annotations to `chiseltest`. The biggest problem would be that the user would have to make sure to only compile to LoFirrtl and not Verilog directly in order to be able to take advantage of the various backends.

Just an idea. Once we have a solid example, we can try out some approaches and think through all the implications together.

mwachs5, 2021-09-09:
I think a concrete example might be rocket-chip's `ElaborationArtefact`s phase: https://github.com/chipsalliance/rocket-chip/blob/86a2f2cca699f149bcc082ef2828654a0a4e3f4b/src/main/scala/stage/phases/GenerateArtefacts.scala

ekiwi, 2021-09-09:
> I think a concrete example might be rocket-chip's `ElaborationArtefact`s phase: https://github.com/chipsalliance/rocket-chip/blob/86a2f2cca699f149bcc082ef2828654a0a4e3f4b/src/main/scala/stage/phases/GenerateArtefacts.scala

Thanks.

Would something like this work for you:

```.scala
val circuit = new RocketChipStage().run(Seq(RunFirrtlTransformAnnotation(new LowFirrtlEmitter), EmitCircuitAnnotation(classOf[LowFirrtlEmitter])))

test(circuit) { dut =>
  dut.in.poke(...)
   ...
}
```

debs-sifive, 2021-09-09:
Yeah, something like that would be great! Do you have an idea of what type `circuit` would be? Like would it be the generated RTL/a RTL generator?

ekiwi, 2021-09-09:
> Do you have an idea of what type `circuit` would be?

It would be an `AnnotationSeq`. We would have to properly define what annotations `chiseltest` requires. Mostly that should be a `TargetDir` annotation and some way to get the firrtl, like a `FirrtlCircuitAnnotation`.
One problem I see is that you might want to filter out the `TargetDir` annotation from the stage. `chiseltest` will automatically create a test directory (for the simulation binary and the resulting VCD) based on the name of your test, but only if there isn't a `TargetDir` annotation already.

ekiwi, 2021-09-15:
@debs-sifive I have been thinking about this a bit more and I have a  proposal.

We would have to add the following class or trait to Chisel (the name isn't the best, happy to change it):

```.scala
trait StageResult[M <: RawModule] {
  def annotations: AnnotationSeq
  def design: M
}
```

Then you would have to change your Stage so that it returns not only the annotations, but also a reference to the Chisel circuit, all packaged up as a `StageResult`. Ideally the stage would have a method like:

```.scala
def run[M <: RawModule](gen: => M, annos: AnnotationSeq): StageResult[M]
```

If you can implement that, we can overload the `test` method in `chiseltest` to accept a `StageResult`.

What do you think? Would this solution work for you? Any downsides that I am missing?

debs-sifive, 2021-09-15:
I think that is a reasonable solution. Internally, we'd probably want to modify a number of our existing custom stages, but it seems like a pretty minor change + worthwhile for this functionality.

Others who have worked with stage stuff — @seldridge @azidar @jackkoenig @mwachs5 — do you have any thoughts on this approach?

mwachs5, 2022-08-24:
Is there not already an annotation that is expected to contain the Scala circuit (top level Module pointer)? If there was, would we really have to change Chisel to accommodate this?

ekiwi, 2022-08-24:
> Is there not already an annotation that is expected to contain the Scala circuit (top level Module pointer)? If there was, would we really have to change Chisel to accommodate this?

The problem is that whatever data gets returned needs to be parameterized on the Circuit type. Since the test function gets a handle to the circuit object, the Scala compiler needs to know the precise type.

closed 2023-01-11 by ekiwi