ucb-bar/chiseltest #398: How to get simulation time

Z-KN, 2021-09-14:
When I am building a testbench, I want to get the current simulation time after some step function, so what is related function in chisel-tester2? In other words, what is the counterpart in chisel-tester2 of $time, a built-in function, in verilog?

ekiwi, 2021-09-14:
Currently we do not supply an interface for this. That could be added, but since `chiseltest` has no notion of real time, the simulation time that we would be able to provide is essentially the number of lock ticks that have elapsed. Would that work for your use-case? You can always multiply the step count with your clock period to get the real time elapsed.

schoeberl, 2021-09-23:
Access to the clock tick counter would make sense for easy handling of timeouts.

ekiwi, 2021-09-23:
> Access to the clock tick counter would make sense for easy handling of timeouts.

Timeouts are already supported through the [setTimeout](https://github.com/ucb-bar/chisel-testers2/blob/a1afe0516a49f296cafb5aeab68b87413e96c2ef/src/test/scala/chiseltest/tests/TimeoutTest.scala) function.

schoeberl, 2021-09-27:
> On 23 Sep, 2021, at 17:37, Kevin Laeufer ***@***.***> wrote:
> 
> 
> Access to the clock tick counter would make sense for easy handling of timeouts.
> 
> Timeouts are already supported through the setTimeout <https://github.com/ucb-bar/chisel-testers2/blob/a1afe0516a49f296cafb5aeab68b87413e96c2ef/src/test/scala/chiseltest/tests/TimeoutTest.scala> function.
> 
> 
OK, fine. But I meant your test bench might like to do something more advanced on a timeout.

Martin

carlosedp, 2021-10-22:
You mean having something like `dut.clock.peek()` that returns the amount of elapsed ticks? I think it might be useful.

ekiwi, 2021-10-22:
> You mean having something like `dut.clock.peek()` that returns the amount of elapsed ticks? I think it might be useful.

Yes. Just that it would be called `getTicks` or something similar since `peek()` already returns whether the clock is high or low.

carlosedp, 2022-03-23:
Implemented in #510

closed 2022-03-23 by ekiwi