ucb-bar/chiseltest #461: chiseltest.TimeoutException: timeout on clock: IO[Clock] at 1000 idle cycles

Lucas-Wye, 2021-12-17:
source code here:
```scala
        for(a <- 1 to 1001) {
          dut.clock.step()
          if(dut.out_valid.peek().litValue == 1) {
            utest.assert(dut.out.peek().litValue == res)
          }
        }
```
The problem is that we can simulate for more than 1000 cycles. Could you give me some solutions?

Lucas-Wye, 2021-12-17:
The source code is: 
`https://github.com/sequencer/arithemic/blob/modmul/arithmetic/tests/src/crypto/MontgomerySpec.scala`
with commit id:
`https://github.com/sequencer/arithemic/commit/a3a5eadf8e469d278d256cad0412377d2bffd2a8`

jackkoenig, 2021-12-17:
This really needs to be in the README, bit you can set the timeout to `0` which disables it, eg.
```scala
dut.clock.setTimeout(0)
```

See tests: https://github.com/ucb-bar/chisel-testers2/blob/a1afe0516a49f296cafb5aeab68b87413e96c2ef/src/test/scala/chiseltest/tests/TimeoutTest.scala#L28

ducky64, 2021-12-17:
Another option might be to include that in the error message.

jackkoenig, 2021-12-17:
> Another option might be to include that in the error message.

Fantastic idea, done in #462

PR #462 (2021-12-17): Provide a suggestion in TimeoutException
Fixes #461 h/t @ducky64 for the idea.
closed 2021-12-17 by mergify[bot]