ucb-bar/chiseltest #466: IndexOutOfBoundsException in witnessToTrace

liuyic00, 2021-12-19:
The error message is as follows. I found it thrown in `chiseltest.formal.backends.Maltese.witnessToTrace`

```log
...
  java.lang.IndexOutOfBoundsException: 1
  at scala.collection.immutable.Vector.checkRangeConvert(Vector.scala:131)
  at scala.collection.immutable.Vector.updateAt(Vector.scala:253)
  at scala.collection.immutable.Vector.updated(Vector.scala:144)
  at chiseltest.formal.backends.Maltese$.$anonfun$expandMemWrites$2(Maltese.scala:153)
  at chiseltest.formal.backends.Maltese$.$anonfun$expandMemWrites$2$adapted(Maltese.scala:149)
  at scala.collection.immutable.List.foreach(List.scala:431)
  at chiseltest.formal.backends.Maltese$.expandMemWrites(Maltese.scala:149)
  at chiseltest.formal.backends.Maltese$.$anonfun$witnessToTrace$8(Maltese.scala:177)
  at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:286)
  at scala.collection.Iterator.foreach(Iterator.scala:943)
...
```

### Steps to Reproduce

Get the code at [this commit](https://github.com/liuyic00/NutShell/commit/a3199e3dd831956b062158a9b866b088795717b5)

```shell
# clone & init submodule
git clone https://github.com/liuyic00/NutShell.git
cd NutShell
git checkout -b bug a3199e3dd831956b062158a9b866b088795717b5
git submodule init
git submodule update 

# run the test
mill __.test
```

### My environment

- Chisel Verions: 3.5-SNAPSHOT
- chiseltest: 0.5-SNAPSHOT
- OS: Darwin Kernel Version 21.2.0: Sun Nov 28 20:28:41 PST 2021; root:xnu-8019.61.5~1/RELEASE_ARM64_T6000 arm64

ekiwi, 2021-12-20:
This might take a while to get fixed. In the meantime, if you switch the backend to `botrmc`, things seem to work.

Best way to get btormc is from here: https://github.com/YosysHQ/oss-cad-suite-build

To switch to that backend, add this annotation to the call to `verify`: 

```.diff
-    verify(new NutCore()(NutCoreConfig()), Seq(BoundedCheck(3)))
+    verify(new NutCore()(NutCoreConfig()), Seq(BoundedCheck(3), BtormcEngineAnnotation))
```

ekiwi, 2022-01-18:
It would be great if you could upgrade your project to the `X.6-SNAPSHOT` releases and see if the recent PR #490 helped with your problem.

closed 2022-01-18 by ekiwi