ucb-bar/chiseltest #473: Verilator version?

schoeberl, 2022-01-04:
Does ChiselTest 0.5.0 need a specific Verilator version? My build fails on a standard Ubuntu box on Travis now. See: https://app.travis-ci.com/github/schoeberl/chisel-book/jobs/554148757

ekiwi, 2022-01-04:
A list of supported Verilator versions is here: https://github.com/ucb-bar/chiseltest#verilator-versions

Can you confirm that your version is `v4.028`?

ekiwi, 2022-01-04:
I need to add some more info to the assertion that your test is tripping. It normally should only fire if the major version is lower than 4 which would be a really old version of verilator. If you can post the output of `verilator --version`, that would help me debug what is going on.

https://github.com/ucb-bar/chiseltest/blob/4d1734ef23490872ee7dfa8fc46aab75c8bc3dc7/src/main/scala/chiseltest/simulator/VerilatorSimulator.scala#L115

schoeberl, 2022-01-04:
Interestingly on my Mac I had yesterday the same problem, but today all Verilator examples run OK. On my Mac I have 4.108. I am not aware that I changed anything on my setup.

But Trevis is a problem. I added the version command and it says:
```
Verilator 3.874 2015-06-06 rev verilator_3_872-20-g0d43051
```
This is what I get from:
```
sudo apt-get install scala texlive-full verilator
```

Maybe I need a newer Ubuntu version to have an actual Verilator version. Will check.

ekiwi, 2022-01-04:
> Maybe I need a newer Ubuntu version to have an actual Verilator version. Will check.

Yes. If you look at the section in the readme that I linked, the oldest version we officially support is the one that ships with Ubuntu 20.04 LTS.

ekiwi, 2022-01-04:
Nowadays I normally recommend people to use Github Actions CI with the oss-cad-tools from Tabby like we do for `ip-contributions`: https://github.com/freechipsproject/ip-contributions/blob/master/.github/workflows/ci.yml
Besides providing a recent Verilator version, it also includes the tools you need for formal verification.

schoeberl, 2022-01-04:
Upgrading Trevis to use Ubuntu 20.04 (and JDK to 11) solves it, as that one installs Verilator version 4.028. If interested, this is my Trevis configuration:
```
dist: focal
before_install:
  - sudo apt-get install scala texlive-full verilator
language: scala
scala:
 - 2.12.13
jdk:
 - openjdk11
script:
 - verilator --version
 - make
```

Now I am green again :-) https://app.travis-ci.com/github/schoeberl/chisel-book

Maybe it is not so important, but Ubuntu 16 is still the default configuration on Trevis and others might be there as well. So a note to the release notes might be worth it.

Yes, I know that GitHub Actions is probably the future and I might switch (sometime). But one step at a time. Until now I was happy with Trevis.

Sorry, if this was too much noise from my side ;-) And thanks for looking into it.

closed 2022-01-04 by schoeberl

ekiwi, 2022-01-04:
No worries, good to know someone is using `chiseltest` :)

HaFred, 2022-01-30:
Hi @schoeberl , I am running one verilator test in chisel-book:
https://github.com/schoeberl/chisel-book/blob/3d396e0e3ecd6d8f91225329b524258c1ac62f16/src/test/scala/BlackBoxesTest.scala#L47
But I ran into errors as below, I wonder do you happen to know how to deal with it? Thanks a lot!
```
Cannot run program "make" (in directory "~/chisel-book"): error=2, No such file or directory
java.io.IOException: Cannot run program "make" (in directory "~/chisel-book"): error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at os.proc.proc$lzycompute$1(ProcessOps.scala:128)
	at os.proc.proc$1(ProcessOps.scala:122)
	at os.proc.spawn(ProcessOps.scala:135)
	at os.proc.call(ProcessOps.scala:76)
	at chiseltest.simulator.VerilatorSimulator$.run(VerilatorSimulator.scala:197)
	at chiseltest.simulator.VerilatorSimulator$.compileSimulation(VerilatorSimulator.scala:153)
	at chiseltest.simulator.VerilatorSimulator$.createContextFromScratch(VerilatorSimulator.scala:124)
	at chiseltest.simulator.VerilatorSimulator$.$anonfun$createContext$1(VerilatorSimulator.scala:69)
	at chiseltest.simulator.Caching$.cacheSimulationBin(Caching.scala:27)
	at chiseltest.simulator.VerilatorSimulator$.createContext(VerilatorSimulator.scala:69)
	at chiseltest.internal.BackendExecutive$.start(BackendExecutive.scala:36)
	at chiseltest.defaults.package$.createDefaultTester(defaults.scala:21)
	at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:31)
	at BlackBoxesTest.$anonfun$new$1(BlackBoxesTest.scala:53)
	at org.scalatest.OutcomeOf.outcomeOf(OutcomeOf.scala:85)
	at org.scalatest.OutcomeOf.outcomeOf$(OutcomeOf.scala:83)
	at org.scalatest.OutcomeOf$.outcomeOf(OutcomeOf.scala:104)
	at org.scalatest.Transformer.apply(Transformer.scala:22)
	at org.scalatest.Transformer.apply(Transformer.scala:20)
	at org.scalatest.flatspec.AnyFlatSpecLike$$anon$5.apply(AnyFlatSpecLike.scala:1683)
	at org.scalatest.TestSuite.withFixture(TestSuite.scala:196)
	at org.scalatest.TestSuite.withFixture$(TestSuite.scala:195)
	at BlackBoxesTest.chiseltest$ChiselScalatestTester$$super$withFixture(BlackBoxesTest.scala:8)
	at chiseltest.ChiselScalatestTester.$anonfun$withFixture$1(ChiselScalatestTester.scala:82)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
	at chiseltest.ChiselScalatestTester.withFixture(ChiselScalatestTester.scala:82)
	at chiseltest.ChiselScalatestTester.withFixture$(ChiselScalatestTester.scala:79)
	at BlackBoxesTest.withFixture(BlackBoxesTest.scala:8)
	at org.scalatest.flatspec.AnyFlatSpecLike.invokeWithFixture$1(AnyFlatSpecLike.scala:1681)
	at org.scalatest.flatspec.AnyFlatSpecLike.$anonfun$runTest$1(AnyFlatSpecLike.scala:1693)
	at org.scalatest.SuperEngine.runTestImpl(Engine.scala:306)
	at org.scalatest.flatspec.AnyFlatSpecLike.runTest(AnyFlatSpecLike.scala:1693)
	at org.scalatest.flatspec.AnyFlatSpecLike.runTest$(AnyFlatSpecLike.scala:1675)
	at org.scalatest.flatspec.AnyFlatSpec.runTest(AnyFlatSpec.scala:1685)
	at org.scalatest.flatspec.AnyFlatSpecLike.$anonfun$runTests$1(AnyFlatSpecLike.scala:1751)
	at org.scalatest.SuperEngine.$anonfun$runTestsInBranch$1(Engine.scala:413)
	at scala.collection.immutable.List.foreach(List.scala:431)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:401)
	at org.scalatest.SuperEngine.runTestsInBranch(Engine.scala:390)
	at org.scalatest.SuperEngine.$anonfun$runTestsInBranch$1(Engine.scala:427)
	at scala.collection.immutable.List.foreach(List.scala:431)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:401)
	at org.scalatest.SuperEngine.runTestsInBranch(Engine.scala:396)
	at org.scalatest.SuperEngine.runTestsImpl(Engine.scala:475)
	at org.scalatest.flatspec.AnyFlatSpecLike.runTests(AnyFlatSpecLike.scala:1751)
	at org.scalatest.flatspec.AnyFlatSpecLike.runTests$(AnyFlatSpecLike.scala:1750)
	at org.scalatest.flatspec.AnyFlatSpec.runTests(AnyFlatSpec.scala:1685)
	at org.scalatest.Suite.run(Suite.scala:1112)
	at org.scalatest.Suite.run$(Suite.scala:1094)
	at org.scalatest.flatspec.AnyFlatSpec.org$scalatest$flatspec$AnyFlatSpecLike$$super$run(AnyFlatSpec.scala:1685)
	at org.scalatest.flatspec.AnyFlatSpecLike.$anonfun$run$1(AnyFlatSpecLike.scala:1796)
	at org.scalatest.SuperEngine.runImpl(Engine.scala:535)
	at org.scalatest.flatspec.AnyFlatSpecLike.run(AnyFlatSpecLike.scala:1796)
	at org.scalatest.flatspec.AnyFlatSpecLike.run$(AnyFlatSpecLike.scala:1794)
	at org.scalatest.flatspec.AnyFlatSpec.run(AnyFlatSpec.scala:1685)
	at org.scalatest.tools.SuiteRunner.run(SuiteRunner.scala:45)
	at org.scalatest.tools.Runner$.$anonfun$doRunRunRunDaDoRunRun$13(Runner.scala:1320)
	at org.scalatest.tools.Runner$.$anonfun$doRunRunRunDaDoRunRun$13$adapted(Runner.scala:1314)
	at scala.collection.immutable.List.foreach(List.scala:431)
	at org.scalatest.tools.Runner$.doRunRunRunDaDoRunRun(Runner.scala:1314)
	at org.scalatest.tools.Runner$.$anonfun$runOptionallyWithPassFailReporter$24(Runner.scala:993)
	at org.scalatest.tools.Runner$.$anonfun$runOptionallyWithPassFailReporter$24$adapted(Runner.scala:971)
	at org.scalatest.tools.Runner$.withClassLoaderAndDispatchReporter(Runner.scala:1480)
	at org.scalatest.tools.Runner$.runOptionallyWithPassFailReporter(Runner.scala:971)
	at org.scalatest.tools.Runner$.run(Runner.scala:798)
	at org.scalatest.tools.Runner.run(Runner.scala)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.runScalaTest2or3(ScalaTestRunner.java:38)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.main(ScalaTestRunner.java:25)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 68 more

```
BTW, I am using verilator 4.028 under ubuntu 20.04.
```bash
ubuntu$ verilator --version
Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1
```

ekiwi, 2022-01-31:
@HaFred can you show us the output of running `make --version`?

HaFred, 2022-01-31:
> @HaFred can you show us the output of running `make --version`?

Ooops, I just realized my make is not working. After reinstallation, everything is ok. Thank you @ekiwi .