ucb-bar/chiseltest #494: Feature request: ChiselTest without ScalaTest

schoeberl, 2022-01-19:
Can we run ChiselTest without ScalaTest? Maybe this is already possible.

One use case is to drive hardware without being a classic test. An example from my digital electronics class: I am providing a Swing-based simulation for some IOs from an FPGA board, so they can play with their vending machine hardware in simulation. This program has a GUI and never terminates by itself. So running it as a test feels wrong. Furthermore, it would block running all other tests with ```sbt test```.

ekiwi, 2022-01-19:
You have two options here:
1. use the [RawTester](https://github.com/ucb-bar/chiseltest/blob/7f731eeb8ebff1521161e091fd5f603cb308b43f/src/main/scala/chiseltest/RawTester.scala#L36)
2. directly use the `chiseltest.simulator` APIs, however, while these low level APIs are a good match for the GUI app you are describing, they are definitely not stable and can be changed on a whim

sequencer, 2022-01-19:
I think `RawTester` works.
However, I think finally a Stage/Phase based Tester should replace original `RawTester`. Then it's possible make different test framework depends on which.

ekiwi, 2022-01-19:
I just thought of a 3rd option and I think this would actually be the best way to make your GUI application: skip the whole chiseltest code and just use treadle directly. Since you are only using this for a class, you probably won't need Verilator support and since you also do not need all the test utility functions, just using treadle directly as a simulator might be your best option.

closed 2022-01-19 by ekiwi

sequencer, 2022-01-19:
There should be a dependency issue for chiseltest to depend on ScalaTest, which makes user not being able to select their own ScalaTest version, and makes backward compatibility tough to maintain/upgrade.

schoeberl, 2022-01-19:
Perfect! I have the simulation as an App back. Still paced in the test folder it simply runs with ```sbt test:run```.

I might look into Treadle at some time, but now I am too lazy to learn a new API ;-)

jurevreca12, 2022-09-29:
I am trying to do a similar thing, where I dynamically simulate a circuit by sending (and receiving) data from python. Currently I am using Treadle directly, but I want to extend this to support also Verilator.  @ekiwi I tried to figure out how to use the chiseltest.simulator API directly, but I couldn't figure it out (I was trying to use the GenericBackend class). Could someone help me with this?

ekiwi, 2022-09-29:
You can find a good example here: https://github.com/ucb-bar/chiseltest/blob/5cc2a8556fd28dc1bff2f5d5b8f55ee7825fd265/src/test/scala/chiseltest/simulator/benchmark/SimulatorBenchmark.scala#L59

However, please note, that there are no API stability guarantees for this API. It might change at any time.

jurevreca12, 2022-09-30:
Thank you for the example. I managed to get it to work.

jurevreca12, 2022-09-30:
I managed to get working an example where I have a "SimulatorContext" object. And I can run simulation by using the notation sim.poke("io_in", BigInt(...)). Is it perhaps also possible to get a similar object as when writing a chiseltest test? Specifically, I want to use DecoupledDriver, but I can't figure out how to achieve this. I see that the object DecoupledDriver is created implicitly from any ReadyValidIO interface object, but the problem is that I can't access the ReadyValidIO object in a "normal" function (i.e. the function that receives data from python).

ekiwi, 2022-09-30:
> I managed to get working an example where I have a "SimulatorContext" object. And I can run simulation by using the notation sim.poke("io_in", BigInt(...)).

Yes. So this API is similar to treadle's API which is why I thought that is what you wanted.

> Is it perhaps also possible to get a similar object as when writing a chiseltest test?

If you want to have more of a `chiseltest` API (with the downside that you will need to have the Chisel code and not only the firrtl) then you can use the [RawTester](https://github.com/ucb-bar/chiseltest/blob/7f731eeb8ebff1521161e091fd5f603cb308b43f/src/main/scala/chiseltest/RawTester.scala#L36)

jurevreca12, 2022-10-02:
I am having trouble understanding how I can use RawTester dynamically. RawTester takes a testing function as an argument, so I am not sure how I can attach this to a simulation running based on what data is received (from python).

ekiwi, 2022-10-02:
> RawTester takes a testing function as an argument

The testing function can execute any arbitrary Scala code.

jurevreca12, 2022-10-03:
(I am forced into this because I am using gRPC for communication, and I am already supplied a function that receives data from python)

ekiwi, 2022-10-03:
Yeah, unfortunately it won't be super pretty ... but it should be doable.

zxmarcos, 2023-05-26:
> You can find a good example here:
>
> https://github.com/ucb-bar/chiseltest/blob/5cc2a8556fd28dc1bff2f5d5b8f55ee7825fd265/src/test/scala/chiseltest/simulator/benchmark/SimulatorBenchmark.scala#L59 [...]

Compiler object is private. How to generate CircuitState from chisel module without it?

ekiwi, 2023-05-26:
> Compiler object is private. How to generate CircuitState from chisel module without it?

Either you place your code in a package that starts with `chiseltest`, or you can copy over the code you need from https://github.com/ucb-bar/chiseltest/blob/8902fe1093458ea21afebbac7507f8fbde80ead1/src/main/scala/chiseltest/simulator/Compiler.scala#L14

zxmarcos, 2023-06-12:
> Either you place your code in a package that starts with `chiseltest`, or you can copy over the code you need from [...]

Is it possible to generate VCD file while using simulator? I tried `val lowFirrtl = Compiler.toLowFirrtl(highFirrtl, List(WriteVcdAnnotation))` with Treadle backend but it does not generate anything.

BTW, I am using simulator api with scala swing for debugging my cpu core, it is really great!