ucb-bar/chiseltest #522: Can't find field of aggregate Memory in nested Module

canoriz, 2022-04-21:
Same bug as [#971](https://github.com/chipsalliance/chisel3/issues/971) and [#968](https://github.com/chipsalliance/chisel3/issues/968) in `chisel3` repo.
```scala
class Top(hexMemoryPath: String) extends Module {
  val io = IO(new Bundle {
    ...
  })
  val core = Module(new Core())
  val memory = Module(new Memory(hexMemoryPath)) // This line cause bug
  ...
}

class Memory(hexMemoryPath: String) extends Module {
  val io = IO(new Bundle {
    ...
  })
  val mem = Mem(0x4000, UInt(8.W))

  loadMemoryFromFile(mem, hexMemoryPath)
  ...
}
```

It compiles, but when testing using
```scala
class Test extends FlatSpec with ChiselScalatestTester {
  behavior of "my code"

  it should "run" in {
    test(new Top(p)) { c =>  // bug
        ...
    }
  }
}
```

Run `sbt test` get error
```
[info] - should src/test/scala/resources/memory/m16k_1.memory *** FAILED ***
[info]   java.util.NoSuchElementException: key not found: Memory.mem
[info]   at scala.collection.MapOps.default(Map.scala:274)
[info]   at scala.collection.MapOps.default$(Map.scala:273)
[info]   at scala.collection.AbstractMap.default(Map.scala:405)
[info]   at scala.collection.mutable.HashMap.apply(HashMap.scala:425)
[info]   at treadle.executable.MemoryInitializer.$anonfun$memoryMetadata$1(Memory.scala:661)
[info]   at scala.collection.immutable.List.flatMap(List.scala:293)
[info]   at scala.collection.immutable.List.flatMap(List.scala:79)
[info]   at treadle.executable.MemoryInitializer.<init>(Memory.scala:654)
[info]   at treadle.executable.ExecutionEngine.<init>(ExecutionEngine.scala:118)
[info]   at treadle.executable.ExecutionEngine$.apply(ExecutionEngine.scala:662)
[info]   at treadle.TreadleTester.<init>(TreadleTester.scala:43)
[info]   at treadle.stage.phases.CreateTester$.transform(CreateTester.scala:19)
[info]   at treadle.stage.phases.CreateTester$.transform(CreateTester.scala:10)
[info]   at treadle.stage.TreadleTesterPhase.$anonfun$transform$1(TreadleTesterPhase.scala:21)
[info]   at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:169)
[info]   at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:165)
[info]   at scala.collection.immutable.List.foldLeft(List.scala:79)
[info]   at treadle.stage.TreadleTesterPhase.transform(TreadleTesterPhase.scala:21)
[info]   at chiseltest.simulator.TreadleSimulator$.createContext(TreadleSimulator.scala:38)
[info]   at chiseltest.internal.BackendExecutive$.start(BackendExecutive.scala:36)
[info]   at chiseltest.defaults.package$.createDefaultTester(defaults.scala:21)
[info]   at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:31)
```

ekiwi, 2022-04-21:
Could you try running the test with Verilator and show the output for that?
Just add `.withAnnotations(Seq(VcsBackendAnnotation))` right after your cal to `test(...)`.

canoriz, 2022-04-22:
It seems I did not install `vcs`. I suppose they are optional and I simply use the default simulator (maybe `treadle`).

Since `vcs` is commercial software, I use `verilator` instead. I add  `.withAnnotations(Seq(VerilatorBackendAnnotation))`. The bug disappears, these tests work as expected.

So is the bug in `treadle`?

ekiwi, 2022-04-22:
> It seems I did not install `vcs`. I suppose they are optional and I simply use the default simulator (maybe `treadle`).

Sorry, I meant to suggest Verilator, got the annotations confused.

> Also, loadMemoryFromFile not work when using Verilator.

How can you tell that it is not working?

> I tried .withAnnotations(Seq(VerilatorBackendAnnotation)), it works. But the previous test without withAnnotation also works even after I remove Verilator. I can't reproduce the bug anymore.

Do you have a reproducible test that I should look at (i.e. a repo that I can download and a sequence of commands that I can run to see the same bug that you are seeing)? Or should we close this issue for now?

canoriz, 2022-04-22:
I have one commit you can take a look and I have a Github Action shows the log.
https://github.com/canoriz/riscv32-cpu/tree/3263395eba6b0e4d5c53fd5c5d788f62976c3ab6
The problem no longer happens, I think we can close the issue.

ekiwi, 2022-04-22:
OK. Let's close this for now. Feel free to re-open if you are bothered by the same issue again.