ucb-bar/chiseltest #528: Extend DecoupledDriver with Enqueue/DequeueEmpty

erlingrj, 2022-05-09:
I am using empty Decoupled interfaces which is just a ready/valid signal to synchronize modules. A very simple and convenient feature would be to have a enqueueEmpty and expectDequeueEmpty methods in DecoupledDriver to drive these interfaces. Or are there a way to do it with the current API?

Thanks

ducky64, 2022-05-09:
What does your Decoupled interface look like? It has to take some data type, right? Even if that type is an empty Bundle or 0-wide integer (not sure if the latter is allowed). Can you call enqueue/dequeue, passing in that data type?

I'd not be a huge fam of having enqueueEmpty or dequeueEmpty because it wouldn't be valid for most Decoupled interfaces, and especially if the compositional approach work, like `myDecoupled.enqueueNow(EmptyBundle().Lit())`.

erlingrj, 2022-05-09:
No, I dont specify any data type. Like this: https://scastie.scala-lang.org/dmjRHxryTYW5Sif9HiJHiA I was actually surprised that you could have an empty Decoupled signal.

ducky64, 2022-05-09:
Ah. So `Decoupled` provides a [default constructor](https://github.com/chipsalliance/chisel3/blob/master/src/main/scala/chisel3/util/Decoupled.scala#L113) that fills in the `EmptyBundle` for you.

For now, see if the above works: `myDecoupled.enqueueNow(new EmptyBundle().Lit())`

In the longer term, since this empty-Decoupled is part of the chisel API, it might make some sense to add support for it in ChiselTest. I'm not sure if we can match on DecoupledIO[EmptyBundle] to define a specialized implicit, or if we want to add something like an empty enqueue that would rely on a runtime check.

erlingrj, 2022-05-09:
Thanks for your help. I tried your suggestion, but `EmptyBundle` is a private class inside Decoupled, so not available. I made my own copy of it, but that gave a "Record type mismatch" error at the "x.bits.poke(data)" statement in the enqueue driver.

ducky64, 2022-05-09:
I see. How about this?

`myDecoupled.enqueueNow(chiselTypeOf(myDecoupled.bits).Lit())`

erlingrj, 2022-05-10:
Thanks again. It turned out the correct syntax was

```
myDecoupled.enqueueNow(chiselTypeOf(myDecoupled.bits))
```

closed 2022-08-15 by erlingrj