ucb-bar/chiseltest #549: FACING SAME ERROR IN DOING SIMPLE ADDITION IN CHISEL

BhattSoham, 2022-08-13:
Hi there,
    Good day. I am new in chisel. I am doing a simple addition and want to see the add value till the expected last value. Can you please help me out in  this code. Actually, it is throwing the same value again and again. I am not able to figure it out where's the error:

 val io = IO(new Bundle {
  
  val a = Input(UInt(8.W))
  val b = Input(UInt(8.W))
  val p = Output(UInt(8.W))
  
})

io.p := io.a + io.b
 
}

test(new ADD) { c =>

val limit = 10
for(i <- 0 until limit) {

val in_A = 5
val in_B = 6
 
 c.io.a.poke(in_A.U)
 c.io.b.poke(in_B.U)
 
 c.clock.step(1)
 
 
 
 println( "Resultx:" + c.io.p.peek() +i)
   
   }
  }
 }
 }

Output:

Resultx:UInt<8>(11)0
Resultx:UInt<8>(11)1
Resultx:UInt<8>(11)2
Resultx:UInt<8>(11)3
Resultx:UInt<8>(11)4
Resultx:UInt<8>(11)5
Resultx:UInt<8>(11)6
Resultx:UInt<8>(11)7
Resultx:UInt<8>(11)8
Resultx:UInt<8>(11)9

Your time and consideration is appreciated.

Thanks & Regards,
Soham Bhattacharya.

ekiwi, 2022-08-13:
> Actually, it is throwing the same value again and again.

You are always adding `5` and `6` and thus the result is always `11`.

closed 2022-08-13 by ekiwi

BhattSoham, 2022-08-13:
Thanks for the reply. Actually, I want to know like if I am passing 5 & 6, I will get the error of 11. I just want to know that, suppose there's two conditions, x = x+ 1 & y = y+2 and I want to work it in a loop. I don't want to add manually every time to the chisel test. i want that to be done with the loop. Is that possible in chisel? But whenever I am working on it, it is giving the same result always. Can you please help me with a code?

To be precise, suppose x = 2, y = 3
So, 1st step : x = x + 1 = 3
 y = y + 2 = 5

The next step which will be x = 3 + 1 = 4 & y == 5 + 2 = 7. I want this in a loop (not adding it up in chisel test manually). Can it be done?

schoeberl, 2022-08-15:
You cannot write x = x + 1 for combinational logic. You need a register in between. HW != SW.

Martin

BhattSoham, 2022-08-15:
Actually, I have used the register. But after two steps, the value is not coming. Basically, when I am using a loop like 

 val io = IO(new Bundle {
  val x_in = Input(FixedPoint(32.W, 16.BP))
  val y_in = Input(FixedPoint(16.W, 8.BP))
  val h = Input(FixedPoint(32.W, 16.BP))
  val y_out = Output(FixedPoint(32.W, 16.BP))
})
val mx = Reg(FixedPoint(32.W, 16.BP))
for(i<-0 until 5) {
mx = io.y_in + io.x_in
}
io.y_out := mx

}

My question is that it is throwing the same result. and the loop is not rolling. But point is just like in C, we have to denote the 'i' in statement within the loop, but in chisel, it is throwing we cannot use it as because it is fixed point. Any idea?