ucb-bar/chiseltest #553: Chiseltest fails with None.get on zero-width literal

hutch31, 2022-08-17:
When running a test with a Bundle with zero-width elements, a fatal None.get error message is reported.

This Scastie reproduces the issue:
https://scastie.scala-lang.org/hutch31/lV9nKmAWRluQ4YTPI2EZ0A/10

@jackkoenig reported that:
I think the issue is that chisel3 or chiseltest is unable to handle unspecified fields in the Bundle literal (DontCare's don't have a litValue which is why you're seeing a None.get). Now because pool is zero-width, you can't specify it because then it doesn't exist and you'll get a different error (an assertion failure that should give a nicer error message)

kammoh, 2022-08-17:
I’ve been facing the sane issue and came up with the following workaround: https://github.com/ucb-bar/chiseltest/pull/552
Implicit/automatic handling of zero-width data would definitely look cleaner, but I don’t know of a clean and reliable way that works across all backends.

ekiwi, 2023-01-11:
This should be fixed now with: https://github.com/ucb-bar/chiseltest/commit/7d98ab58459832607361d045a90bdbccfe265297

closed 2023-01-11 by ekiwi