ucb-bar/chiseltest #567: Working Verilator Version

stevenmburns, 2022-12-02:
Which version of verilator should I be using with chiseltest?

It seems that v4.106 works, and so does v4.220, but v4.202 and v.5.002 do not.

Should we continue to use these specific older versions? Is this documented correctly somewhere? This table in the README.md is incorrect about v4.202

[image]

ekiwi, 2022-12-02:
`v4.202` seems to work in our continuous integration testing: https://github.com/ucb-bar/chiseltest/actions/runs/3257195383/jobs/5348508976

Is our CI broken in a sense that we use the wrong Verilator version or are we just missing a test case?

ekiwi, 2022-12-02:
The goal would be to support all new Verilator versions. However that might require some help from the community as I currently do not have the time to implement support for newer versions.

ekiwi, 2022-12-02:
The most up-to date list of supported versions is here: https://github.com/ucb-bar/chiseltest/blob/0894e8fba74674bcd32e7070211b40eb5be1b20b/.github/workflows/test.yml#L99, however my goal is to always keep the README in sync.

stevenmburns, 2022-12-03:
It seems that those failing versions are only on our internal servers at Intel. They work fine with an Ubuntu 18.04 build.
I'll track down this issue on my end. Good to know you are trying to support even the newest versions of Verilator.

closed 2022-12-03 by stevenmburns