ucb-bar/chiseltest #605: Regression with FixedPoint poke in 0.5.1+

nrother, 2023-01-10:
I noticed a regression introduced in chiseltest 0.5.1 (it is still present in 0.5.5). The following test case will work on 0.5.0, but fails (incorrectly) on higher versions.

```scala
import org.scalatest._
import chisel3._
import chiseltest._
import chisel3.util._
import chisel3.experimental.FixedPoint
import math.Pi

class PassModule extends Module
{
  val io = IO(new Bundle {
    val in = Input(FixedPoint(16.W, 15.BP))
    val out = Output(UInt(16.W))
  })

  io.out := io.in.asUInt()
}

class MiscTest extends flatspec.AnyFlatSpec with ChiselScalatestTester
{
  behavior of "ChiselTest"

  it should "correctly poke FixedPoint values" in {
    test(new PassModule()) { dut =>
      val in = (-13.86568775047758717754).F(15.BP)

      dut.io.in.poke(in)
      dut.clock.step()
      val out = dut.io.out.peek().litValue

      //this fails in chiseltest 0.5.1+, producing 4395 (==0x112B)
      assert(out == 4401) //4401 == 0x1131
    }
  }
}
```

This does only happen for certain FixedPoint values, and seems to be related to the truncation (small values do not produce any problems). Let me know if you need any further information.

ekiwi, 2023-03-03:
This is a real bug, unfortunately I do not have time to fix it and upstream Chisel decided to remove FixedPoint support, so we are going to remove it as well: https://github.com/ucb-bar/chiseltest/pull/624

ekiwi, 2023-06-27:
Unfortunately we no longer support FixedPoint.

closed 2023-06-27 by ekiwi