ucb-bar/chiseltest #613: any way seperate  compile and simulation

pxs7, 2023-01-13:
in some testbench,  just simulation args need be change to run different case， but just need compile once, is there any way seprate chiseltest compile and simulation

ekiwi, 2023-01-13:
Currently this is not supported, but this feature has been requested in the past. If you want to make it more likely that this will eventually be implemented it would be great if you could write down a description of your use-case. Just tell us a bit about your design, how many tests you think could use the same simulation binary and how much time you estimate you could save.

pxs7, 2023-01-13:
thanks， current no many case， just thinking out the simulating pross of vcs， which seprate compile and simulating

closed 2023-01-13 by pxs7