ucb-bar/chiseltest #621: Function loadMemoryFromFile failed in chiseltest

B10615053, 2023-02-20:
I had a CPU Module which contains an Instruction Memory Module.
```scala=
class CPU_Datapath_Memory extends Module 
{
    val im = Module(new InstMem(15))
```
The Instruction Memory Module uses preloaded memory that loads `inst.hex` with `loadMemoryFromFile` function.
```scala=
import chisel3._
import chisel3.util._
import chisel3.util.experimental.loadMemoryFromFile
import firrtl.annotations.MemoryLoadFileType

class InstMem(bits:Int) extends Module 
{
	val io = IO(new Bundle 
	{
		//32kB -> bits = 15
		val raddr = Input(UInt(bits.W))
		val inst = Output(UInt(32.W))
	})
	val memory = Mem((1<<bits), UInt(8.W))
	loadMemoryFromFile(memory, "./inst.hex")

	val rdata = Wire(UInt(32.W))
	rdata := Cat(memory((io.raddr+3.U)),
				memory((io.raddr+2.U)),
				memory((io.raddr+1.U)),
				memory((io.raddr+0.U)))

	io.inst :=rdata
}
```
At first, I tested it with `chisel3.iotesters`, and the tester works perfectly.
```scala=
import chisel3.iotesters.{PeekPokeTester,Driver}
import scala.language.implicitConversions

object CPU_Datapath_Memory_Test_Work extends App
{
    Driver.execute(args,()=>new CPU_Datapath_Memory)
    {
        c:CPU_Datapath_Memory => new CPU_Datapath_Memory_Test_Work(c)
    }
}
```
After that, I rewrote the tester with `chiseltest`, but `loadMemoryFromFile` function did not load  `inst.hex` at all as waveform shown below.
```scala=
import chisel3._
import chiseltest._
import org.scalatest.flatspec.AnyFlatSpec

class CPU_Datapath_Memory_Test_Fail extends AnyFlatSpec with ChiselScalatestTester
{
    "Test CPU" should "PASS" in 
    {
        test(new CPU_Datapath_Memory).withAnnotations(Seq(VerilatorBackendAnnotation, WriteVcdAnnotation))
        {
        dut =>
```
[image]
[image]
Am I using `loadMemoryFromFile` function wrongly, or is there another way to preload memory in `chiseltest`?

ekiwi, 2023-02-20:
Thanks for reporting a bug. Could you let us know what version of `chiseltest` you are using? It should be in your `build.sbt`.

B10615053, 2023-02-21:
My running environment are listed below:

+ Operating System: Ubuntu 20.04.5 LTS
+ Java: openjdk version 11.0.17 2022-10-18
+ Chisel: 3.5.5
+ ChiselTest: 0.5.4