ucb-bar/chiseltest #627: Confusing Fork-Join Usage

liuxd17thu, 2023-03-28:
I am often confused about how to use fork and join in chiseltest.
Are there any good advice(or examples) about understanding how they work? 
Here's an example showing my problem:
https://gist.github.com/liuxd17thu/c2bf5880cf605545e1835908ce60c46e

Besides, how do I know what causes "overlapping pokes" from the error message?

ducky64, 2023-04-06:
Can you post the error message / stack trace?

Overlapping pokes probably means two threads tried to poke the same line

liuxd17thu, 2023-04-07:
Sure. If I only keep the B part and comment out A and C [like this](https://gist.github.com/liuxd17thu/c2bf5880cf605545e1835908ce60c46e)
```Scala
// ... Line 36
        fork{
          d.io.enq1.enqueueSeq(Seq(10.U, 11.U, 12.U))
        }.fork{
//        --- A ---
//          d.clock.step(10)
//          d.io.deq1.expectDequeueSeq(Seq(10.U, 11.U, 12.U)) // This works fine, of course.
//        --- B ---
//        What should I do if I have to avoid using `expectXXX()` methods? ( since I cannot know the exact output in some complex modules other than Queue )
//        This will give me a timeout error. `deq1.ready` keeps false in waveform.
          d.clock.step(10)
          d.io.deq1.ready.poke(true.B)
//        --- C ---
//        This pokes `deq1.ready` for only 1 cycle
//        d.clock.step(10)
//        fork{
//          d.io.deq1.ready.poke(true.B)
//          d.clock.step(1)
//        }.join
        }.join
// Line 54 ...
```
Then I get:
```
timeout on dualQueue.clock: IO[Clock] at 1000 idle cycles. You can extend the timeout by calling .setTimeout(<n>) on your clock (setting it to 0 means 'no timeout').
chiseltest.TimeoutException: timeout on dualQueue.clock: IO[Clock] at 1000 idle cycles. You can extend the timeout by calling .setTimeout(<n>) on your clock (setting it to 0 means 'no timeout').
	at chiseltest.internal.GenericBackend.$anonfun$run$7(GenericBackend.scala:203)
	at chiseltest.internal.GenericBackend.$anonfun$run$7$adapted(GenericBackend.scala:200)
	at scala.collection.mutable.HashMap.$anonfun$foreach$1(HashMap.scala:149)
	at scala.collection.mutable.HashTable.foreachEntry(HashTable.scala:237)
	at scala.collection.mutable.HashTable.foreachEntry$(HashTable.scala:230)
	at scala.collection.mutable.HashMap.foreachEntry(HashMap.scala:44)
	at scala.collection.mutable.HashMap.foreach(HashMap.scala:149)
	at chiseltest.internal.GenericBackend.run(GenericBackend.scala:200)
	at chiseltest.internal.Context$.$anonfun$run$1(Testers2.scala:34)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
	at chiseltest.internal.Context$.run(Testers2.scala:34)
	at chiseltest.ChiselScalatestTester.chiseltest$ChiselScalatestTester$$runTest(ChiselScalatestTester.scala:103)
	at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:31)
	at play.simple_test.$anonfun$new$1(NaiveTest.scala:25)
	at org.scalatest.OutcomeOf.outcomeOf(OutcomeOf.scala:85)
	at org.scalatest.OutcomeOf.outcomeOf$(OutcomeOf.scala:83)
	at org.scalatest.OutcomeOf$.outcomeOf(OutcomeOf.scala:104)
	at org.scalatest.Transformer.apply(Transformer.scala:22)
	at org.scalatest.Transformer.apply(Transformer.scala:20)
	at org.scalatest.flatspec.AnyFlatSpecLike$$anon$5.apply(AnyFlatSpecLike.scala:1683)
	at org.scalatest.TestSuite.withFixture(TestSuite.scala:196)
	at org.scalatest.TestSuite.withFixture$(TestSuite.scala:195)
	at play.simple_test.chiseltest$ChiselScalatestTester$$super$withFixture(NaiveTest.scala:8)
	at chiseltest.ChiselScalatestTester.$anonfun$withFixture$1(ChiselScalatestTester.scala:82)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
	at chiseltest.ChiselScalatestTester.withFixture(ChiselScalatestTester.scala:82)
	at chiseltest.ChiselScalatestTester.withFixture$(ChiselScalatestTester.scala:79)
	at play.simple_test.withFixture(NaiveTest.scala:8)
	at org.scalatest.flatspec.AnyFlatSpecLike.invokeWithFixture$1(AnyFlatSpecLike.scala:1681)
	at org.scalatest.flatspec.AnyFlatSpecLike.$anonfun$runTest$1(AnyFlatSpecLike.scala:1693)
	at org.scalatest.SuperEngine.runTestImpl(Engine.scala:306)
	at org.scalatest.flatspec.AnyFlatSpecLike.runTest(AnyFlatSpecLike.scala:1693)
	at org.scalatest.flatspec.AnyFlatSpecLike.runTest$(AnyFlatSpecLike.scala:1675)
	at org.scalatest.flatspec.AnyFlatSpec.runTest(AnyFlatSpec.scala:1685)
	at org.scalatest.flatspec.AnyFlatSpecLike.$anonfun$runTests$1(AnyFlatSpecLike.scala:1751)
	at org.scalatest.SuperEngine.$anonfun$runTestsInBranch$1(Engine.scala:413)
	at scala.collection.immutable.List.foreach(List.scala:431)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:401)
	at org.scalatest.SuperEngine.runTestsInBranch(Engine.scala:390)
	at org.scalatest.SuperEngine.$anonfun$runTestsInBranch$1(Engine.scala:427)
	at scala.collection.immutable.List.foreach(List.scala:431)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:401)
[... 3 lines cut ...]
	at org.scalatest.flatspec.AnyFlatSpecLike.runTests$(AnyFlatSpecLike.scala:1750)
	at org.scalatest.flatspec.AnyFlatSpec.runTests(AnyFlatSpec.scala:1685)
	at org.scalatest.Suite.run(Suite.scala:1112)
	at org.scalatest.Suite.run$(Suite.scala:1094)
	at org.scalatest.flatspec.AnyFlatSpec.org$scalatest$flatspec$AnyFlatSpecLike$$super$run(AnyFlatSpec.scala:1685)
	at org.scalatest.flatspec.AnyFlatSpecLike.$anonfun$run$1(AnyFlatSpecLike.scala:1796)
	at org.scalatest.SuperEngine.runImpl(Engine.scala:535)
	at org.scalatest.flatspec.AnyFlatSpecLike.run(AnyFlatSpecLike.scala:1796)
	at org.scalatest.flatspec.AnyFlatSpecLike.run$(AnyFlatSpecLike.scala:1794)
	at org.scalatest.flatspec.AnyFlatSpec.run(AnyFlatSpec.scala:1685)
	at org.scalatest.tools.SuiteRunner.run(SuiteRunner.scala:45)
	at org.scalatest.tools.Runner$.$anonfun$doRunRunRunDaDoRunRun$13(Runner.scala:1320)
	at org.scalatest.tools.Runner$.$anonfun$doRunRunRunDaDoRunRun$13$adapted(Runner.scala:1314)
	at scala.collection.immutable.List.foreach(List.scala:431)
	at org.scalatest.tools.Runner$.doRunRunRunDaDoRunRun(Runner.scala:1314)
[... cut ...]
	at org.scalatest.tools.Runner$.runOptionallyWithPassFailReporter(Runner.scala:971)
	at org.scalatest.tools.Runner$.run(Runner.scala:798)
	at org.scalatest.tools.Runner.run(Runner.scala)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.runScalaTest2or3(ScalaTestRunner.java:43)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.main(ScalaTestRunner.java:26)
```
And `io.deq1.ready` keeps 0 in the vcd waveform. Am I missing something important about `fork` usage?