ucb-bar/chiseltest #644: scala.NotImplementedError: TODO: convet InlineAnnotation(ModuleName(<omitted>))

yupferris, 2023-06-13:
Hi, I'm migrating a project from chisel `3.6.0` to `5.0.0`, along with chiseltest `5.0-SNAPSHOT`, which both appear to resolve correctly.

However, when running tests, I hit the following exception:

```
[info]   scala.NotImplementedError: TODO: convert InlineAnnotation(ModuleName(<omitted>))
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:127)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$annosToState$2(ChiselBridge.scala:87)
[info]   at scala.collection.immutable.List.map(List.scala:246)
[info]   at scala.collection.immutable.List.map(List.scala:79)
[info]   at chiseltest.simulator.ChiselBridge$.annosToState(ChiselBridge.scala:87)
[info]   at chiseltest.simulator.ChiselBridge$.elaborate(ChiselBridge.scala:67)
[info]   at chiseltest.simulator.Compiler$.elaborate(Compiler.scala:15)
[info]   at chiseltest.internal.BackendExecutive$.start(BackendExecutive.scala:19)
[info]   at chiseltest.defaults.package$.createDefaultTester(defaults.scala:21)
```

I can look into making a minimal (shareable) reproduction, but in the meantime, maybe the above is enough?

Cheers!

closed 2023-06-14 by ekiwi

yupferris, 2023-06-15:
Thanks for getting to this so quickly!

Should I expect the snapshot to have been updated? I'm still seeing the same errors as above. I've tried `sbt update`, `sbt clean test`, removing various cache directories, etc. Any tips?

ekiwi, 2023-06-15:
CI published the new snapshot: https://github.com/ucb-bar/chiseltest/actions/runs/5270395633/jobs/9530014044
Not sure why you aren't seeing it. For me, the directory I normally need to delete is `~/.cache/coursier/v1/https/`.

yupferris, 2023-06-15:
Aha, indeed that's a directory I missed, and it looks like that did the trick. Cheers, and thanks again!

ekiwi, 2023-06-15:
Great to hear that this worked. Can you let me know if you get all your tests working with the new chiseltest? Once that is the case I might try to do a stable 5.0.0 release for chiseltest.

yupferris, 2023-06-16:
They're all confirmed working :)

yupferris, 2023-06-20:
@ekiwi pinging in case you didn't see my above comment.

ekiwi, 2023-06-20:
> @ekiwi pinging in case you didn't see my above comment. Nothing needed on my end; just making sure you saw it :)

I saw it. Thanks!

I might have time to try to do a release this week. But please do not depend on it.

yupferris, 2023-06-26:
Forgot to respond sooner, but wanted to say that's great and no rush, we're not dependent on any such release schedule. Thanks!

ekiwi, 2023-06-27:
There is a chiseltest release now: https://github.com/ucb-bar/chiseltest/releases/tag/v5.0.0

yupferris, 2023-06-27:
> There is a chiseltest release now: https://github.com/ucb-bar/chiseltest/releases/tag/v5.0.0

Thanks for the heads-up!

yupferris, 2023-06-28:
Just wanted to let you know the new release appears to be working without a hitch, cheers :)