ucb-bar/chiseltest #649: JRE detects `EXCEPTION_ACCESS_VIOLATION` when trying to use Verilator as Chiseltest's backend

hiroshi-ya, 2023-06-25:
When trying to use Verilator as Chiseltest's backend, I got the following error messages:
```
E:\MyCode\Research\chisel\pipelinedMult> sbt "testOnly multTest"
[info] welcome to sbt 1.8.2 (Eclipse Adoptium Java 11.0.19)
[info] loading global plugins from C:\Users\storm\.sbt\1.0\plugins
[info] loading project definition from E:\MyCode\Research\chisel\pipelinedMult\project
[info] loading settings for project pipelinedmult from build.sbt ...
[info] set current project to pipelinedmult (in build file:/E:/MyCode/Research/chisel/pipelinedMult/)
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ffcb6b32b1d, pid=9936, tid=27808
#
# JRE version: OpenJDK Runtime Environment Temurin-11.0.19+7 (11.0.19+7) (build 11.0.19+7)
# Java VM: OpenJDK 64-Bit Server VM Temurin-11.0.19+7 (11.0.19+7, mixed mode, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# V  [jvm.dll+0x2f2b1d]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# E:\MyCode\Research\chisel\pipelinedMult\hs_err_pid9936.log
#
# If you would like to submit a bug report, please visit:
#   https://github.com/adoptium/adoptium-support/issues
#
``` 
My OS is Windows 11. I'm using `Verilator v4.202` (via MSYS2), `Chisel3 v3.6.0`, and `Chiseltest v0.6.0`. I have no trouble using Chisel to generate Verilog files, nor using Treadle as Chiseltest's backend. However, I'm struggling to get Chiseltest using Verilator on Windows. 

The full error report [hs_err_pid9936.log](https://github.com/ucb-bar/chiseltest/files/11859516/hs_err_pid9936.log) is here. Any advice would be deeply appreciated.

ekiwi, 2023-06-27:
Looks like the JNI component that we use to load Verilator simulations does not work properly on your Windows system.
Unfortunately Windows support (especially for the native simulators) is somewhat "best effort" and mostly relies on Windows users debugging problems and submitting bug reports. Personally I do not have the experience nor time to debug Windows problems.

ekiwi, 2023-06-27:
This was the last such PR that seemed to make chiseltest work on windows for the author: https://github.com/ucb-bar/chiseltest/pull/507