ucb-bar/chiseltest #664: Solver Chosen Constants for Formal Verification

fayalalebrun, 2023-08-06:
In SymbiYosys, the `anyconst` attribute can be used in order to have the solver choose a constant for an input variable. For certain types of designs, this can make proving correctness easier.

Is there currently a way to specify such an attribute with chiseltest?

ekiwi, 2023-08-06:
An uninitialized register should essentially be able to emulate the behavior of `anyconst`.:

```.scala
val myConst = Reg(UInt(8.W))
```

Registers without a reset value will start at an arbitrary constant (solver chosen in case of formal) and then, assuming you do not assign anything to them, they will retain their value.

fayalalebrun, 2023-08-06:
Would the following have the same effect?

```
assume(stable(input))
```

ekiwi, 2023-08-06:
> Would the following have the same effect?

I think so. The only downside here is that to make an input you need to be in the top-level module. Registers can easily declared independent of module hierarchy.