ucb-bar/chiseltest #682: scala.NotImplementedError: TODO: convert ThrowOnFirstErrorAnnotation

FJShen, 2023-09-13:
**Software version I'm using**: `org.chipsalliance::chisel:5.0.0`, `edu.berkeley.cs::chiseltest:5.0.0`, `org.chipsalliance:::chisel-plugin:5.0.0` running on `scala 2.13.8`.
**How to reproduce error**: Write up a ChiselTest like this:
```
class Datapath_test extends AnyFreeSpec with ChiselScalatestTester{

  "blah blah blah" in {
    test(new Datapath).withChiselAnnotations(
      Seq(ThrowOnFirstErrorAnnotation)
    ){dut => ... }
```
**Error message**:
```
- blah blah blah *** FAILED ***                              
  scala.NotImplementedError: TODO: convert ThrowOnFirstErrorAnnotation        
  at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:149)        
  at chiseltest.simulator.ChiselBridge$.$anonfun$annosToState$2(ChiselBridge.scala:93)
  at scala.collection.immutable.List.map(List.scala:246)                                  
  at scala.collection.immutable.List.map(List.scala:79)                     
  at chiseltest.simulator.ChiselBridge$.annosToState(ChiselBridge.scala:93)    
  at chiseltest.simulator.ChiselBridge$.elaborate(ChiselBridge.scala:73)               
  at chiseltest.simulator.Compiler$.elaborate(Compiler.scala:19)                                      
  at chiseltest.internal.BackendExecutive$.start(BackendExecutive.scala:23)
  at chiseltest.defaults.package$.createDefaultTester(defaults.scala:25)       
  at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:32)                                                                                                                      
  ...                                                                                        
```

**Other information**:
specifying `PrintFullStackTraceAnnotation` instead of `ThrowOnFirstErrorAnnotation` also yields the similar "NotImplementedError".

ekiwi, 2023-09-16:
Thanks! This should be fixed in the latest `6.0-SNAPSHOT` release.
I am also working on getting the fix into a `5.0.2` release.

closed 2023-09-16 by ekiwi

ekiwi, 2023-09-18:
@FJShen here you go: https://github.com/ucb-bar/chiseltest/releases/tag/v5.0.2

FJShen, 2023-09-18:
Thanks a lot!

PR #685 (2023-09-16): ChiselBridge: filter out chisel options
fixes #682