ucb-bar/chiseltest #697: scala.NotImplementedError: TODO: convert DecodeTableAnnotatio

Discreater, 2023-12-31:
Hi, I'm upgrading my project from chisel 3.6 to 5.1, along with chiseltest `5.0.2`.
But when I run the tests, I get an error.
```
TODO: convert DecodeTableAnnotation(~...)
scala.NotImplementedError: TODO: convert DecodeTableAnnotation(~...)
	at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:151)
	at chiseltest.simulator.ChiselBridge$.$anonfun$annosToState$2(ChiselBridge.scala:93)
	at scala.collection.immutable.List.flatMap(List.scala:293)
	at scala.collection.immutable.List.flatMap(List.scala:79)
	at chiseltest.simulator.ChiselBridge$.annosToState(ChiselBridge.scala:93)
	at chiseltest.simulator.ChiselBridge$.elaborate(ChiselBridge.scala:73)
	at chiseltest.simulator.Compiler$.elaborate(Compiler.scala:19)
	at chiseltest.internal.BackendExecutive$.start(BackendExecutive.scala:23)
	at chiseltest.defaults.package$.createDefaultTester(defaults.scala:25)
	at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:32)
...
```
I'm using `chisel3.util.experimental.decode.{TruthTable, decoder}` in my code, is this enough information and do I need to provide reproducible code?

ekiwi, 2023-12-31:
This is enough info. Can you bump me on January 15th on this issue? Before that I won't be able to take care of it.

Discreater, 2023-12-31:
OK, thanks!

ekiwi, 2023-12-31:
If you have some time and want to help, you could try to PR a small regressions test which uses the `chisel3.util.experimental.decode.{TruthTable, decoder}` API. That would make it easier for me to test the fix.

Discreater, 2024-01-02:
This looks a bit odd, but is it just a matter of adding the corresponding `Annotation` in firrtl2? I gave it a try, and it seems to work with the tests. If so, I can make a PR.

ekiwi, 2024-01-02:
> This looks a bit odd, but is it just a matter of adding the corresponding `Annotation` in firrtl2? I gave it a try, and it seems to work with the tests. If so, I can make a PR.

Sounds about right. I would be excited to review a PR if you want to make one.

closed 2024-01-03 by ekiwi

Gianthard-cyh, 2024-01-06:
I'm facing the same issue when using chiseltest on this module:
```scala
import chisel3._
import chisel3.util._
import chisel3.util.experimental.decode._
import chisel3.util.BitPat
import InstType._

class CtrlSignal extends Bundle {
  // GPR Register File Control Signals
  val GPRAddrA   = UInt(5.W)
  val GPRAddrB   = UInt(5.W)
  val GPRAddrW   = UInt(5.W)
  val GPRWen     = Bool()
  val GPRWSrcSel = UInt(1.W)
  // ALU Control Signals
  val ALUFnCode    = UInt(3.W)
  val ALUInputSelA = UInt(1.W)
  val ALUInputSelB = UInt(1.W)
  // PCR Control Signals
  val PCRSrcSel = UInt(1.W)
}

class ImmGen extends Module {
  val io = IO(new Bundle {
    val inst     = Input(UInt(32.W))
    val instType = Input(UInt(3.W))
    val imm      = Output(SInt(32.W))
  })
  val immR = 0.S(32.W)
  val immI = Cat(io.inst(31, 20), 0.U(20.W)).asSInt
  val immS = Cat(io.inst(31, 25), io.inst(11, 7)).asSInt
  val immB = Cat(io.inst(31), io.inst(7), io.inst(30, 25), io.inst(11, 8)).asSInt
  val immU = io.inst(31, 12).asSInt
  val immJ = Cat(io.inst(31), io.inst(19, 12), io.inst(20), io.inst(30, 21)).asSInt

  io.imm := MuxLookup(io.instType, 0.S(32.W))(
    Seq(
      ("b" + R).U -> immR,
      ("b" + I).U -> immI,
      ("b" + S).U -> immS,
      ("b" + B).U -> immB,
      ("b" + U).U -> immU,
      ("b" + J).U -> immJ
    )
  )
}

object InstParse extends DecodeField[InstPattern, UInt] {
  def chiselType: UInt = UInt(8.W)
  def name = "Instruction control signal handling"
  def genTable(i: InstPattern) = BitPat("b" + i.ctrlSig + i.opALU)
}

object InstTypeFetch extends DecodeField[InstPattern, UInt] {
  def chiselType: UInt = UInt(3.W)
  def name = "Inst type fetch"
  def genTable(i: InstPattern) = BitPat("b" + i.instType)
}

class IDU extends Module {
  val io = IO(new Bundle {
    val inst       = Input(UInt(32.W))
    val ctrlSignal = Output(new CtrlSignal)
    val imm        = Output(SInt(32.W))
  })

  val instLUT = Seq(
    InstPattern("????????????_?????_000_?????_0010011", "addi", I, "000", "10000"),
    InstPattern("????????????_?????_001_?????_0010011", "subi", I, "011", "10000")
  )

  val immGen = Module(new ImmGen())
  io.ctrlSignal.GPRAddrA := io.inst(19, 15)
  io.ctrlSignal.GPRAddrB := io.inst(24, 20)
  io.ctrlSignal.GPRAddrW := io.inst(11, 7)

  val decodeTable  = new DecodeTable(instLUT, Seq(InstParse, InstTypeFetch))
  val decodeResult = decodeTable.decode(io.inst)

  immGen.io.instType := decodeResult(InstTypeFetch)
  immGen.io.inst     := io.inst
  io.imm             := immGen.io.imm

  io.ctrlSignal.GPRWen     := decodeResult(InstParse)(7)
  io.ctrlSignal.GPRWSrcSel := decodeResult(InstParse)(6)

  io.ctrlSignal.ALUInputSelA := decodeResult(InstParse)(5)
  io.ctrlSignal.ALUInputSelB := decodeResult(InstParse)(4)

  io.ctrlSignal.PCRSrcSel := decodeResult(InstParse)(3)

  io.ctrlSignal.ALUFnCode := decodeResult(InstParse)(2, 0)
}
```
I'm facing this issue when using `DecodeTable`. Hope the code I posted here will help you to get more information about this. @ekiwi

ekiwi, 2024-01-06:
@Discreater submitted a patch (#698 ) which is now part of the latest `6.0-SNAPSHOT` release of `chiseltest`. Does that work for you?

PR #698 (2024-01-03): Add convert DecodeTableAnnotation regression test
Closes #697.

The `Booth` module and test is from https://github.com/sequencer/arithmetic/blob/master/arithmetic/src/multiplier/Booth.scala.

Feel free to modify anything.

- ~~Waiting for https://github.com/ucb-bar/firrtl2/pull/29 to be merged~~ (Not necessary)