ucb-bar/chiseltest #699: Will there be a chiseltest 6.0.0?

schoeberl, 2024-01-09:
The verification story of *new* Chisel (CIRCT) needs more hands to move forward.

ekiwi, 2024-01-09:
There currently is a `6.0-SNAPSHOT`. Once Chisel 6 is released and I have some free time, I was planning on taking whatever we have at that point in `6.0-SNAPSHOT` and putting it into a `6.0.0` release.

schoeberl, 2024-01-09:
Great! I know that chiseltest (6.0) is not using CIRCT, but it will keep testing alive as it is. We need some more manpower for the verification side on *new* Chisel.

gonsolo, 2024-01-20:
As I said in https://github.com/chipsalliance/chisel/issues/3746 I hope this gets resolved. Also my hope is that in the future chisel 7 would get released *together* with chiseltest 7 and chisel-template 7!

kammoh, 2024-02-09:
@gonsolo 
Could you please give the latest `6.0-SNAPSHOT` a try? #705 should probably have resolved that issue.

ezhes, 2024-02-11:
Is `6.0-SNAPSHOT` released somewhere? Not sure how to actually get it, doesn't appear in any package repositories:

```
[error] stack trace is suppressed; run last update for the full output
[error] (update) sbt.librarymanagement.ResolveException: Error downloading edu.berkeley.cs:chiseltest_2.13:6.0-SNAPSHOT
[error]   Not found
[error]   Not found
[error]   not found: /home/xxx/.ivy2/localedu.berkeley.cs/chiseltest_2.13/6.0-SNAPSHOT/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/edu/berkeley/cs/chiseltest_2.13/6.0-SNAPSHOT/chiseltest_2.13-6.0-SNAPSHOT.pom
[error] Total time: 3 s, completed Feb 10, 2024, 6:47:10 PM
```

ekiwi, 2024-02-11:
You need to have the snapshot resolver enabled in your build.sbt:
```
resolvers ++= Resolver.sonatypeOssRepos("snapshots"),
```

I am still planning to release  6.0, just have been pretty busy recently.

rameloni, 2024-02-28:
> There currently is a `6.0-SNAPSHOT`. Once Chisel 6 is released and I have some free time, I was planning on taking whatever we have at that point in `6.0-SNAPSHOT` and putting it into a `6.0.0` release.
> 
> _Note_: `chiseltest` `6.0.0` won't be using CIRCT though which will limit access to some of the newer features.

@ekiwi so, will chiseltest 6.0.0 keep using the same elaboration steps of chiseltest 5? SFC instead of firtool?

gonsolo, 2024-02-28:
This might be of interest: https://github.com/chipsalliance/chisel/issues/3746#issuecomment-1967542565

rameloni, 2024-02-28:
> This might be of interest: [chipsalliance/chisel#3746 (comment)](https://github.com/chipsalliance/chisel/issues/3746#issuecomment-1967542565)

@gonsolo I see, then what cases are users supposed to use chiseltest instead of chiselsim? What are the advantages of using chiseltest?

gonsolo, 2024-02-28:
> @gonsolo I see. Then what cases are users supposed to use chiseltest instead of chiselsim? What are the advantages of using chiseltest?

None. My (fairly basic) understanding is this: chiseltest will go away, you have svsim which is based on verilator, and now ChiselSim on top of svsim which provides a similar experience as chiseltest (peek & poke, more advanced things like fork/join maybe in the making).
  
> However, this is something pretty recent, it was published yesterday.

Let's see how it's going to evolve.

rameloni, 2024-02-28:
> eek & poke, more advanced things like fork/join maybe in the making).

I quickly checked it and I don't see also the support for waveform emission.

gonsolo, 2024-02-28:
> I quickly checked it and I don't see also the support for waveform emission.

You can do this in Verilator, I guess it's just a matter of adding some support for that in ChiselSim and svsim.

Again, I just glanced over this yesterday and I might be wrong.

rameloni, 2024-02-28:
I think I got it, it seems that `EphemeralSimulator` is just a super simple lightweight simulator that just executes verilator.

ekiwi, 2024-02-28:
Here we go! Finally a `6.0.0` release: https://github.com/ucb-bar/chiseltest/releases/tag/v6.0.0

closed 2024-02-28 by ekiwi