ucb-bar/chiseltest #707: Constant exceptions on macOS with the threaded Verilator backend

kammoh, 2024-02-01:
I'm not able to run many of the testbenches which worked just fine with any version of chiseltest since the new threaded scheduler was merged in.

The structure of my testbenches are quiet simple:
```scala
for( testcase <- allTestCases){
      fork {
        dut.io.cmd.enqueue(command)
      }.fork {
        dut.io.din.enqueueSeq(inputWords)
      }.fork {
        dut.io.dout.expectDequeueSeq(expectedWords)
      }.join() // or joinAndStep()
}
```
The module under test (`dut`) has some `Decoupled` input and output interfaces like the above code.

The number testcases is in order of multiple 1000s.

Exceptions I get:

-  "Cannot terminate thread waiting for join"
```
  java.lang.AssertionError: assertion failed: Cannot terminate thread waiting for join chiseltest_thread_85 (85)
  at scala.Predef$.assert(Predef.scala:279)
  at chiseltest.internal.Scheduler.chiseltest$internal$Scheduler$$terminateAllChildThreads(Scheduler.scala:427)
  at chiseltest.internal.Scheduler.finishMainThread(Scheduler.scala:446)
  at chiseltest.internal.SimController.run(SimController.scala:127)
  at chiseltest.internal.Context$.$anonfun$runTest$2(Context.scala:30)
  at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
  at chiseltest.internal.Context$.runTest(Context.scala:30)
  at chiseltest.ChiselScalatestTester.chiseltest$ChiselScalatestTester$$runTest(ChiselScalatestTester.scala:96)
  at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:30
```

- Deadlock
```
[E] Exception in thread "chiseltest_thread_4204" java.lang.RuntimeException: Deadlock!
[E]     at chiseltest.internal.Scheduler.$anonfun$chiseltest$internal$Scheduler$$findNextThread$2(Scheduler.scala:307)
[E]     at scala.Option.getOrElse(Option.scala:201)
[E]     at chiseltest.internal.Scheduler.chiseltest$internal$Scheduler$$findNextThread(Scheduler.scala:304)
[E]     at chiseltest.internal.Scheduler.chiseltest$internal$Scheduler$$finishThread(Scheduler.scala:200)
[E]     at chiseltest.internal.Scheduler.$anonfun$createThread$1(Scheduler.scala:148)
[E]     at java.base/java.lang.Thread.run(Thread.java:842)
[E] Exception in thread "chiseltest_thread_9663" java.lang.AssertionError: assertion failed: 9663: R
[E]     at scala.Predef$.assert(Predef.scala:279)
[E]     at chiseltest.internal.Scheduler.chiseltest$internal$Scheduler$$finishThread(Scheduler.scala:180)
[E]     at chiseltest.internal.Scheduler.$anonfun$createThread$1(Scheduler.scala:148)
[E]     at java.base/java.lang.Thread.run(Thread.java:842)
```

The exceptions are sparodic and rarely happen with shorter running tests. Nevertheless in testbenches with large number of testcases it happens more often than not! (In some of the testbenches it's almost always).

I'm using the Verilator beckend. Trying with Treadle backed I haven't yet seen the exceptions but I haven't been able to finish any of the tests either (I assume it's just treadle being very slow).

I've tried JDK versions 11, 17, and 21; few different chisel/chiseltest versions; and a couple of different Verilator versions with no luck.

I'm running macOS 14.2.1 on Apple silicon (arm64).

ekiwi, 2024-02-01:
Thanks for reporting these issues.

Could you please set `EnableDebug` in the `Scheduler` class to `true` and then post the output of the problems you are observing? That might make it possible for me to debug them.

kammoh, 2024-02-05:
```
0@0: forkThread(chiseltest_thread_1 (1) P0) from 0
0@0: yieldForFork()
0@0:   --> 1: P(0), [0]: R
0@0:   --> nextThreadId = 1
1@0: forkThread(chiseltest_thread_2 (2) P1) from 1
1@0: yieldForFork()
1@0:   --> [1]: R, 0: P(0), 2: P(0)
1@0:   --> nextThreadId = 1
1@0: joinThreads(ids = List(2))
1@0:   --> nextThreadId = 0
0@0: forkThread(chiseltest_thread_3 (3) P0) from 0
0@0: yieldForFork()
0@0:   --> 1: J(2), 3: P(0), [0]: R, 2: P(0)
0@0:   --> nextThreadId = 3
3@0: forkThread(chiseltest_thread_4 (4) P1) from 3
3@0: yieldForFork()
3@0:   --> 1: J(2), [3]: R, 0: P(0), 2: P(0), 4: P(0)
3@0:   --> nextThreadId = 3
3@0: joinThreads(ids = List(4))
3@0:   --> nextThreadId = 0
0@0: forkThread(chiseltest_thread_5 (5) P0) from 0
0@0: yieldForFork()
0@0:   --> 1: J(2), 3: J(4), 5: P(0), [0]: R, 2: P(0), 4: P(0)
0@0:   --> nextThreadId = 5
5@0: forkThread(chiseltest_thread_6 (6) P1) from 5
5@0: yieldForFork()
5@0:   --> 1: J(2), 3: J(4), [5]: R, 0: P(0), 2: P(0), 4: P(0), 6: P(0)
5@0:   --> nextThreadId = 5
5@0: joinThreads(ids = List(6))
5@0:   --> nextThreadId = 0
0@0: joinThreads(ids = List(1, 3, 5))
0@0:   --> nextThreadId = 2
2@0: finishThread(id=2 (chiseltest_thread_2))
2@0:   --> 1: J(2), 3: J(4), 5: J(6), 0: J(1), 4: P(0), 6: P(0)
1@0: stepThread(cycles = 1): minPause = 0
1@0:   --> [1]: R, 3: J(4), 5: J(6), 0: J(1), 4: P(0), 6: P(0)
1@0: yieldForStep(cycles = 1)
1@0:   --> [1]: P(1), 3: J(4), 5: J(6), 0: J(1), 4: P(0), 6: P(0)
1@0:   --> nextThreadId = 4
4@0: stepThread(cycles = 1): minPause = 0
4@0:   --> 1: P(1), 3: J(4), 5: J(6), 0: J(1), [4]: R, 6: P(0)
4@0: yieldForStep(cycles = 1)
4@0:   --> 1: P(1), 3: J(4), 5: J(6), 0: J(1), [4]: P(1), 6: P(0)
4@0:   --> nextThreadId = 6
6@0: stepThread(cycles = 1): minPause = 1
6@0:   --> 1: P(1), 3: J(4), 5: J(6), 0: J(1), 4: P(1), [6]: R
6@0: doStep(cycles = 1)
6@1: yieldForStep(cycles = 0)
6@1:   --> 1: P(0), 3: J(4), 5: J(6), 0: J(1), 4: P(0), [6]: P(0)
6@1:   --> nextThreadId = 1
1@1: finishThread(id=1 (chiseltest_thread_1))
1@1:   --> 3: J(4), 5: J(6), 0: J(1), 4: P(0), 6: P(0)
0@1:   --> nextThreadId = 4
4@1: finishThread(id=4 (chiseltest_thread_4))
4@1:   --> 3: J(4), 5: J(6), 0: J(3), 6: P(0)
3@1: stepThread(cycles = 1): minPause = 0
3@1:   --> [3]: R, 5: J(6), 0: J(3), 6: P(0)
3@1: yieldForStep(cycles = 1)
3@1:   --> [3]: P(1), 5: J(6), 0: J(3), 6: P(0)
3@1:   --> nextThreadId = 6
6@1: stepThread(cycles = 1): minPause = 1
6@1:   --> 3: P(1), 5: J(6), 0: J(3), [6]: R
6@1: doStep(cycles = 1)
6@2: yieldForStep(cycles = 0)
6@2:   --> 3: P(0), 5: J(6), 0: J(3), [6]: P(0)
6@2:   --> nextThreadId = 3
3@2: forkThread(chiseltest_thread_7 (7) P1) from 3
3@2: yieldForFork()
3@2:   --> [3]: R, 5: J(6), 0: J(3), 7: P(0), 6: P(0)
[... 96 lines cut ...]
0@31: forkThread(chiseltest_thread_15 (15) P0) from 0
0@31: yieldForFork()
0@31:   --> 11: J(12), 13: J(14), 15: P(0), [0]: R, 12: P(0), 14: P(0)
0@31:   --> nextThreadId = 15
15@31: Caught exception (chiseltest.ThreadOrderDependentException). Shutting down and propagating exception to parent.
15@31: finishThread(id=15 (chiseltest_thread_15))
15@31:   --> 11: J(12), 13: J(14), 0: P(0), 12: P(0), 14: P(0)
15@31:   --> nextThreadId = 0
0@31: terminating: chiseltest_thread_11 (11), chiseltest_thread_13 (13), chiseltest_thread_14 (14), chiseltest_thread_12 (12)
- should work *** FAILED ***
  java.lang.AssertionError: assertion failed: Cannot terminate thread waiting for join chiseltest_thread_11 (11)
  at scala.Predef$.assert(Predef.scala:279)
  at chiseltest.internal.Scheduler.chiseltest$internal$Scheduler$$terminateAllChildThreads(Scheduler.scala:427)
  at chiseltest.internal.Scheduler.finishMainThread(Scheduler.scala:446)
  at chiseltest.internal.SimController.run(SimController.scala:127)
  at chiseltest.internal.Context$.$anonfun$runTest$2(Context.scala:30)
  at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
  at chiseltest.internal.Context$.runTest(Context.scala:30)
  at chiseltest.ChiselScalatestTester.chiseltest$ChiselScalatestTester$$runTest(ChiselScalatestTester.scala:96)
  at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:30)
  at keccak.KeccakCoreSpec.$anonfun$new$1(KeccakCoreSpec.scala:36)
  ...
Execution took 2.15s
1 tests, 1 failed
```

kammoh, 2024-02-05:
Here's a minimimal testcase which shows the issue:

```scala
import chisel3._
import chisel3.experimental.BundleLiterals._
import chisel3.util._
import chiseltest._
import org.scalatest.flatspec.AnyFlatSpec

class Dut extends Module {
  val io = IO(new Bundle {
    val cmd = Flipped(Decoupled(new Bundle {
      val op = UInt(3.W)
    }))
    val din = Flipped(Decoupled(UInt(8.W)))
    val dout = Decoupled(UInt(256.W))
  })

  object State extends ChiselEnum {
    val Idle, Receive, Send = Value
  }
  val state = RegInit(State.Idle)

  val counter = Counter(100)
  val accumulator = Reg(io.din.bits.cloneType)

  io.cmd.ready := state === State.Idle
  io.din.ready := state === State.Receive
  io.dout.valid := state === State.Send
  io.dout.bits := accumulator

  when(io.cmd.fire) {
    state := State.Receive
    accumulator := 0.U
  }
  when(io.din.fire) {
    accumulator := accumulator + io.din.bits
    when(counter.inc()) {
      state := State.Send
    }
  }
  when(io.dout.fire) {
    state := State.Idle
  }

}

class MWE extends AnyFlatSpec with ChiselScalatestTester {

  behavior of "MWE"

  val annos = Seq(
    VerilatorBackendAnnotation,
  )

  val rand = scala.util.Random

  it should "work" in {
    test(new Dut()).withAnnotations(annos) { dut =>
      val dw = dut.io.din.bits.getWidth
      for (
        (op, input) <- Seq(
          0.U -> Seq.fill(1000)(BigInt(dw, rand).U(dw.W)),
          1.U -> Seq.fill(2000)(BigInt(dw, rand).U(dw.W)),
          2.U -> Seq.fill(3000)(BigInt(dw, rand).U(dw.W)),
        )
      ) {
        fork {
          dut.io.cmd.enqueue(chiselTypeOf(dut.io.cmd.bits).Lit(_.op -> op))
        }.fork {
          dut.io.din.enqueueSeq(input)
        }.fork {
          dut.io.dout.expectDequeueSeq(Seq((input.map(_.litValue).sum).U(dw.W) ))
        }.join()
      }
    }
  }
}
```

```zsh
❯ bloop test mwe.test --only MWE

MWE:
MWE
0@0: forkThread(chiseltest_thread_1 (1) P0) from 0
0@0: yieldForFork()
0@0:   --> 1: P(0), [0]: R
0@0:   --> nextThreadId = 1
1@0: forkThread(chiseltest_thread_2 (2) P1) from 1
1@0: yieldForFork()
1@0:   --> [1]: R, 0: P(0), 2: P(0)
1@0:   --> nextThreadId = 1
1@0: joinThreads(ids = List(2))
1@0:   --> nextThreadId = 0
0@0: forkThread(chiseltest_thread_3 (3) P0) from 0
0@0: yieldForFork()
0@0:   --> 1: J(2), 3: P(0), [0]: R, 2: P(0)
0@0:   --> nextThreadId = 3
3@0: forkThread(chiseltest_thread_4 (4) P1) from 3
3@0: yieldForFork()
3@0:   --> 1: J(2), [3]: R, 0: P(0), 2: P(0), 4: P(0)
3@0:   --> nextThreadId = 3
3@0: joinThreads(ids = List(4))
3@0:   --> nextThreadId = 0
0@0: forkThread(chiseltest_thread_5 (5) P0) from 0
0@0: yieldForFork()
0@0:   --> 1: J(2), 3: J(4), 5: P(0), [0]: R, 2: P(0), 4: P(0)
0@0:   --> nextThreadId = 5
5@0: Caught exception (java.lang.IllegalArgumentException). Shutting down and propagating exception to parent.
5@0: finishThread(id=5 (chiseltest_thread_5))
5@0:   --> 1: J(2), 3: J(4), 0: P(0), 2: P(0), 4: P(0)
5@0:   --> nextThreadId = 0
0@0: terminating: chiseltest_thread_1 (1), chiseltest_thread_3 (3), chiseltest_thread_4 (4), chiseltest_thread_2 (2)
- should work *** FAILED ***
  java.lang.AssertionError: assertion failed: Cannot terminate thread waiting for join chiseltest_thread_1 (1)
  at scala.Predef$.assert(Predef.scala:279)
  at chiseltest.internal.Scheduler.chiseltest$internal$Scheduler$$terminateAllChildThreads(Scheduler.scala:427)
  at chiseltest.internal.Scheduler.finishMainThread(Scheduler.scala:446)
  at chiseltest.internal.SimController.run(SimController.scala:127)
  at chiseltest.internal.Context$.$anonfun$runTest$2(Context.scala:30)
  at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
  at chiseltest.internal.Context$.runTest(Context.scala:30)
  at chiseltest.ChiselScalatestTester.chiseltest$ChiselScalatestTester$$runTest(ChiselScalatestTester.scala:96)
  at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:30)
  at MWE.$anonfun$new$18(mwe.scala:56)
  ...
Execution took 2.12s
1 tests, 1 failed

===============================================
Total duration: 2.12s
1 failed

Failed:
- MWE:
  * MWE should work - java.lang.AssertionError: assertion failed: Cannot terminate thread waiting for join chiseltest_thread_1 (1)
===============================================

kammoh, 2024-02-05:
I fixed an issue with the minimal example and simplified it a bit. I also enabled `DebugThreadSwitches` in addition to `EnableDebug` in the generated output. Please let me know if there's anything else I can do to help with the debugging.
Thanks!

kammoh, 2024-02-06:
I reinstalled JDK on my system and did a full clean of the project and _ miraculously_ everything has started to work again! I'm no longer able to reproduce the issue, neither with the minimal example or other testbenches!
I had already tried tried the same steps before without any luch so no idea what happened or what was the real cause of the exceptions.
I'm going to close this issue for now and will re-open if they show up again.

closed 2024-02-06 by kammoh

ekiwi, 2024-02-07:
Glad it works for you! Let me know if this happens again.