ucb-bar/chiseltest #718: [Help]A TLRAM test failed log

SpinEch0, 2024-02-22:
Hi, all,

I write a TLRAM test case, below is err log. I dont know why it happened. Can anyone help ?

```
- should run *** FAILED ***
  java.lang.ArrayIndexOutOfBoundsException: Index 1342236047 out of bounds for length 844878
  at treadle2.executable.DataStore$AssignInt.runLean(DataStore.scala:137)
  at treadle2.executable.DataStore$AssignInt.$anonfun$setLeanMode$1(DataStore.scala:149)
  at treadle2.executable.Scheduler.executeAssigners(Scheduler.scala:121)
  at treadle2.executable.Scheduler.executeOrphanedAssigns(Scheduler.scala:135)
  at treadle2.executable.ExecutionEngine.<init>(ExecutionEngine.scala:104)
  at treadle2.executable.ExecutionEngine$.apply(ExecutionEngine.scala:629)
  at treadle2.TreadleTester.<init>(TreadleTester.scala:38)
  at treadle2.stage.phases.CreateTester$.transform(CreateTester.scala:19)
  at treadle2.stage.phases.CreateTester$.transform(CreateTester.scala:10)
  at treadle2.stage.TreadleTesterPhase.$anonfun$transform$1(TreadleTesterPhase.scala:20)
  ...

```

Thanks!

ekiwi, 2024-02-22:
Thanks for the bug report! Can you share some more information?
1. Does this test work when you use the Verilator backend? (add `.withAnnotations(Seq(VerilatorBackendAnnotation))` after your `test(...)`
2. Did this test work with a prior version of `chiseltest`? I.e. is it a regression?
3. How large is the memory in your TLRam?
4. Can you share a repository and instructions to run the test?

SpinEch0, 2024-02-23:
> Thanks for the bug report! Can you share some more information?
>
> 1. Does this test work when you use the Verilator backend? (add `.withAnnotations(Seq(VerilatorBackendAnnotation))` after your [...]

Thanks for your reply.
1. compile failed:
```
test_run_dir/Multi_level_DCache_should_run/verilated/VSimTop_123: undefined symbol: _ZN7VSimTopC1EPKc
```
2. No, because of dependencies limitation.
3. 4GB
4. 
below is my test codes, it depends on https://github.com/OpenXiangShan/XiangShan:
```
import chisel3._
import utility.{TLClientsMerger, ChiselDB, FileRegisters, TLLogger}
import huancun._
import org.chipsalliance.cde.config._
import freechips.rocketchip.diplomacy._
import freechips.rocketchip.tilelink._
import chiseltest._
import chiseltest.simulator.{VerilatorCFlags, VerilatorLinkFlags}
import org.scalatest.flatspec.AnyFlatSpec
import top.DefaultConfig
import xiangshan.cache._
import xiangshan._
import system._
import java.nio.file.Paths

// XiangShan log / perf ctrl, should be inited in SimTop IO
// If not needed, just ingore these signals
class LogPerfIO extends Bundle {
  val timer = UInt(64.W)
  val logEnable = Bool()
  val clean = Bool()
  val dump = Bool()
}

class TestTop_L2()(implicit p: Parameters) extends LazyModule {

  /* L1D   L1D
   *  \    /
   *    L2
   */

  override lazy val desiredName: String = "SimTop"
  val delayFactor = 0.5
  val cacheParams = p(HCCacheParamsKey)

  val dcaches = (0 until 2) map(_ => LazyModule(new DCache()))
  val l1d_nodes = (0 until 2) map( i => dcaches(i).clientNode)
  val l1d_l2_tllog_nodes = (0 until 2) map(i => TLLogger(s"L1D_L2_$i"))

  val l2 = LazyModule(new HuanCun())
  val xbar = TLXbar()
  val ram = LazyModule(new TLRAM(AddressSet(0, 0xffffffffL), beatBytes = 32))

  println(l1d_nodes(0))
  for(i <- 0 until 2) {
    xbar :=* l1d_l2_tllog_nodes(i) := TLBuffer() := l1d_nodes(i)
  }

  ram.node :=
    TLXbar() :=*
      TLFragmenter(32, 64) :=*
      TLCacheCork() :=*
      TLDelayer(delayFactor) :=*
      l2.node :=* xbar

  lazy val module = new Impl
  class Impl extends LazyModuleImp(this) {
    val io = IO(new Bundle {
      val perfInfo = new LogPerfIO
      val dcacheIO = Vec(2, new DCacheIO)
    })

    io.perfInfo := 0.U.asTypeOf(new LogPerfIO)

    val timer = WireDefault(0.U(64.W))
    val logEnable = WireDefault(false.B)
    val clean = WireDefault(io.perfInfo.clean)
    val dump = WireDefault(io.perfInfo.dump)

    dontTouch(timer)
    dontTouch(logEnable)
    dontTouch(clean)
    dontTouch(dump)

    (0 until 2) map(i => io.dcacheIO(i) <> dcaches(i).module.io)

  }
}

class MultiLevelDCacheTest extends AnyFlatSpec with ChiselScalatestTester {
  behavior of "Multi level DCache"

  it should ("run") in {
    val defaultConfig = (new DefaultConfig)
    implicit val config = defaultConfig.alter((_, _, up) => {
      case XSCoreParamsKey => defaultConfig(XSTileKey).head.copy()
      case SoCParamsKey =>
        up(SoCParamsKey).copy(
          PAddrBits = 32)

      case DebugOptionsKey => up(DebugOptionsKey).copy(EnableConstantin = false)
      case HCCacheParamsKey => HCCacheParameters(
        name = "L2",
        level = 2,
        ways = 4,
        sets = 256,
        inclusive = false,
        alwaysReleaseData = true,
        clientCaches = Seq(CacheParameters("dcache", sets = 256, ways = 4, blockGranularity = 5)),
        reqField = Seq(PreferCacheField()),
        echoField = Seq(DirtyField()),
        respKey = Seq(IsHitKey),
        prefetch = Some(huancun.prefetch.BOPParameters()),
        sramDepthDiv = 2,
        simulation = true
      )
    })
    println("========== Test the correctness of syntactic and datapath ==========")

    ChiselDB.init(false)
    val top = LazyModule(new TestTop_L2)

    val cur_path = Paths.get("").toAbsolutePath.toString
    val target_path = Paths.get(cur_path).resolve("./build").toString
    println(s"target path ${target_path}")
    ChiselDB.addToFileRegisters
    FileRegisters.write(fileDir = target_path)

   //  test(top.module).withAnnotations(Seq(VerilatorBackendAnnotation, VerilatorCFlags(Seq("-I${target_path}")), VerilatorLinkFlags(Seq(s"${target_path}/chisel_db.o")))) {
   test(top.module) {
      c =>
        println("========== in test ==========")
        // s0
        c.io.dcacheIO(0).hartId.poke(0.U)
        println("cycle 0!!!!!!!!!!!!!!!!!!")
        c.clock.step(1)
        println("cycle 1!!!!!!!!!!!!!!!!!!")
    }
    println("========== end test ==========")
  }

}

```

ekiwi, 2024-02-23:
> 3\. 4GB

OK, this just sounds like `treadle` might not work with memories this big. If you want to investigate and maybe look into fixing this issue, I would recommend creating a simpler testcase, i.e. a very simple circuit that instantiates a 4GiB memory just like your TLRam does and then try to test that with treadle. Once you can get it to crash, you now have a simple enough test to look into what is missing in treadle to support such large memories.

SpinEch0, 2024-02-26:
> OK, this just sounds like `treadle` might not work with memories this big. If you want to investigate and maybe look into fixing this issue, I would recommend creating a simpler testcase, i.e. a [...]

Test passed with 1GB TLRAM. I see comments in DataStore.scala, it really doesnt support 4GB ram :
```
/** Creates a data store for the three underlying data types. The meaning of the values of each slot must be maintained
  * outside of this class. This class only supports (2 ** 31) - 1 of any ints, longs or bigs.
  */
```
I will close this Issue.

ekiwi, 2024-02-26:
> Test passed with 1GB TLRAM. I see comments in DataStore.scala, it really doesnt support 4GB ram :

Makes sense. Ideally it should throw an exception if we are trying to allocate too much memory, warning you of that fact. Please let me know if you would be interested in making a PR to add such an exception. (You do not need to define your own exception classe, just `throw new RuntimeException(...)`

schoeberl, 2024-02-26:
I guess this is a restriction of the JVM (and not treadle). As far as I remember, arrays in the JVM are indexed with 32-bit integers. If on-chip memory is represented as an array of bytes, this is then maximum 2 GB.

SpinEch0, 2024-02-27:
> > I guess this is a restriction of the JVM (and not treadle). As far as I remember, arrays in the JVM are indexed with 32-bit integers. If on-chip memory is represented as an array of bytes, this is then maximum 2 GB.
> 
> Yes, with the current representation that treadle uses (i.e. storing all values in a flat array), this is a JVM limit. If someone wants to try it, it would make a lot of sense to use a different storage format for memories. We probably want to have an array of references to memory objects. These memory objects could then support dense as well as sparse arrays. A sparse representation with a hashmap would make a lot of sense in the case of a 4GiB array.

Sorry I dont know how to test with chiseltest source code.  I find that chiseltest has MemReg unit test, it throws exception when mem is oversize.
```
sbt:chiseltest> testOnly treadle2.MemRegTester
[info] compiling 1 Scala source to /home/zsj/workspace/chiseltest/target/scala-2.13/classes ...
[info] MemRegTester:
[info] MemReg
[info] - should update registers correctly *** FAILED ***
[info]   treadle2.executable.TreadleException: Memory file size 4294967294 is too large for treadle
[info]   at treadle2.executable.Memory$.buildSymbols(Memory.scala:57)
```

So I guess my case is not because of mem oversize. It may because of below codes:
[executable code](https://github.com/ucb-bar/chiseltest/blob/06e6ea8e7b9cf4952bcafc7c5a727e1edb5d11ba/src/main/scala/treadle2/executable/ExecutionEngine.scala#L598C1-L604C1)
```
599     val dataStoreAllocator = new DataStoreAllocator
600 
601     symbolTable.allocateData(dataStoreAllocator)
602 
603     val dataStore = DataStore(dataStoreAllocator)
604 

// In allocateData, the hashmap value is not verified that it must be not overflow
// Sorry I dont know how to verify it

class DataStoreAllocator {
  val nextIndexFor = new mutable.HashMap[DataSize, Int]

  nextIndexFor(IntSize) = 0
  nextIndexFor(LongSize) = 0
  nextIndexFor(BigSize) = 0

  def numberOfInts:  Int = nextIndexFor(IntSize)
  def numberOfLongs: Int = nextIndexFor(LongSize)
  def numberOfBigs:  Int = nextIndexFor(BigSize)

  val watchList: mutable.HashSet[Symbol] = new mutable.HashSet()

  def getSizes: (Int, Int, Int) = {
    (nextIndexFor(IntSize), nextIndexFor(LongSize), nextIndexFor(BigSize))
  }
  // make sure size is not overflow
  def getIndex(dataSize: DataSize, slots: Int = 1): Int = {
    val index = nextIndexFor(dataSize)
    nextIndexFor(dataSize) += slots
    index
  } 
} 

// when use intData, the size maybe wrong

  def numberOfInts:  Int = dataStoreAllocator.nextIndexFor(IntSize)
  def numberOfLongs: Int = dataStoreAllocator.nextIndexFor(LongSize)
  def numberOfBigs:  Int = dataStoreAllocator.nextIndexFor(BigSize)

  val watchList: mutable.HashSet[Symbol] = new mutable.HashSet()

  val intData:  Array[Int] = Array.fill(numberOfInts)(0)
  val longData: Array[Long] = Array.fill(numberOfLongs)(0L)
  val bigData:  Array[Big] = Array.fill(numberOfBigs)(Big(0))

```

closed 2024-03-01 by SpinEch0