ucb-bar/chiseltest #722: What are the future usecases of ChiselTest if it is replaced by ChiselSim?

rameloni, 2024-02-28:
I have seen that a post about ChiselSim has been recently published. It says that ChiselTest is not longer supported for new chisel features.

So, what are the future developments for chiseltest?

ekiwi, 2024-02-28:
> So, what are the future developments for chiseltest?

Whatever the community wants to do. For now, chiseltest has more features, but there are no paid full-time developers working on it. So in the long term it will be hard to keep up with Chisel development which has almost 10 engineers paid by SiFive working on it.

ekiwi, 2024-02-29:
@rameloni : I saw your issue on whether GTKWave supports plugins. You might want to look into Surfer instead: https://gitlab.com/surfer-project/surfer

The code base is much newer and easier to extend. It already has special support for displaying Spade types (Spade is a new hardware language from the same author as Surfer) and I am sure they would be open to adding support for displaying Chisel and maybe also Tydi types.

rameloni, 2024-02-29:
> @rameloni : I saw your issue on whether GTKWave supports plugins. You might want to look into Surfer instead: https://gitlab.com/surfer-project/surfer
>
> The code base is much newer and easier to [...]

@ekiwi thank you! I will take a look into that, and probably I will ask for opinions from the creator of surfers. 
If you read the issue on gtkwave you'll probably know what my project consists of.
I'm starting now with the actual implementation for the project and the waveform visualisation is something that will come later (I guess in a month). I first need to map and elaborate chisel structures to map them to simulation values.

I also had in mind to create something that will be simulator independent and working with both chiseltest and chiselsim.

ekiwi, 2024-02-29:
> I also had in mind to create something that will be simulator independent and working with both chiseltest and chiselsim.

Honestly, I would just pick one. Think of your thesis as creating a prototype that demonstrates a concept. Otherwise you will get bogged down in too many engineering-y things that won't matter in the end. Feel free to send an email to my @berkeley.edu email if you want to have a chat about your project. It sounds quite interesting and I might have a lot of knowledge that could help you get started.

closed 2024-03-03 by ekiwi