ucb-bar/chiseltest #729: one step takes extremely long time to complete

hukangha, 2024-04-24:
Hi,
Basically I'm testing a circuit with chiseltest, code is like:
`
import chiseltest._
import chiseltest.iotesters._
import org.scalatest.flatspec.AnyFlatSpec

class VCMeshNetworkSanityTest_new extends AnyFlatSpec with ChiselScalatestTester{

  behavior of "VCMeshNetwork"
    it should "behave normally" in {
      test(new VCMeshNetwork(256,Seq(2))(4,4)).withAnnotations(Seq(WriteVcdAnnotation)) { c =>
        val outReadyVC = 2.toLong
        ........
        var timelimit = 100
        var timeout = false
        var cycle = 0
        while((!timeout) && !c.io.port(3)(3).flow_o.valid.peek().litToBoolean){
          step(1)
          cycle +=1
          timeout = cycle>=timelimit
          println(s"cycles: ${cycle}")
        }
...
`
the println is added there to show the progress of the while loop, and the println takes so long to print out the cycles,  and the total time is like 9 minutes for one test(roughly 30 cycles), seems maybe the step(1) or the peek operation is taking a lot of time.
When using the peekpoketester compatibility api, It works faster, and the test executes about 38 seconds.
the compatibility code is something like this:
`  while((!timeout) && (peek(c.io.port(3)(3).flow_o.valid)==0)){
    step(1)
    cycle +=1
    timeout = cycle>=timelimit
  }
`
I have just migrated to chisel6.2.0 and chiseltest 6.0.0. 
not sure what to do.
Any suggestions?

ekiwi, 2024-04-24:
> When using the peekpoketester compatibility api, It works faster, and the test executes about 38 seconds.

That is very surprising. I would generally expect both APIs to perform about the same. Honestly, if the old API is that much faster I would just stick with that.

One thing that you could try, with either the old or the new API , would be to use Verilator in the backend. Just add a `VerilatorBackendAnnotation` to your `.withAnnotations(Seq(...`

hukangha, 2024-04-25:
thank you, my fault not enabled verilator backend