ucb-bar/chiseltest #732: Cover is not yet implemented for Chiseltest

Gallagator, 2024-06-17:
Cover statements are not yet possible. Something like:

```scala
class Foo() extends Module {
  val out = IO(Output(Bool()))
  val i = Counter(10)
  out := i === 9
  cover(out)
}
```

verified with:

```scala
verify((new Foo), CVC4EngineAnnotation, CoverCheck(10))
```
Which should generate a trace showing that it is possible for out to go high.

ekiwi, 2024-06-17:
Thanks for filing this issue. This feature is indeed missing.

ekiwi, 2024-06-28:
My advice would be to pre-process the firrtl. I.e., write a firrtl pass that removes all assert and assume statements and changes all cover statements to asserts. That should be a relatively simple transform. You might be inspired by: https://github.com/ucb-bar/firrtl2/blob/main/src/main/scala/firrtl2/transforms/formal/RemoveVerificationStatements.scala